subsurface/subsurface #2919: Pedix AI - Downloading of dives via bluetooth LE on Linux

ryanm101, 2020-08-19:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Two issues:
1. I must remove from host computer and readd before i can download dives
2. When downloading dives Subsurface downloads the dives from the computer as soon as it is connected (which is very slow). It then asks which dives to import. The shearwater desktop gets a list of dives from the computer and only downloads the ones requested by the user. This behaviour might be preferable in SS as it would be quicker on computers with lots of dives. 

### Operating system:
Linux Fedora 32 - Dell XPS

### Subsurface version:
4.9.6

### Steps to reproduce:
1st Time
1. Pair Perdix AI with OS and look for 'connected'
2. Open Import dives, select vendor/model
3. Tick choose Bluetooth mode, click ...
4. Choose LE Mode, Scan, select discovered device
5. Save
6. Click Download this can fail so keep retrying and it should succeed. 

2nd time:
1. Remove device from host computer
2. Follow steps above and it works

### Current behavior:
1. If you do not remove from host computer 1st it will 'fail' to open/read the device.
2. All dives are downloaded from computer which is slow then i select which to import

### Expected behavior:
1. I should not have to remove and readd to host each time i want to download.
2. Device should list dives and once i select which to import only those should be downloaded. 

### Additional information:

dirkhh, 2020-09-09:
1. agreed - I would expect that with firmware v81 from Shearwater this issue should go away
2. since we support dozens of different dive computer, our ability to customize the app for a specific vendor's setup is limited. I agree that for those vendors that have a directory of dives that could be downloaded / parsed first, it would be nice to do that and then offer to download a specific subset of dives. But that is simply not how things are implemented today. Instead we download all dives that are newer than the newest dive that you have downloaded from THIS dive computer. So if you have 100 dives on your Perdix AI that are synced with Subsurface and you do two new dives, we only download three dives (yes, that's one of the mis-features caused by what I explained above... we do download that newest dive that's already there a second time and once we realize that you already have it, we stop). But we do NOT download all 102 dives.

dirkhh, 2020-09-10:
Yes, the release notes say Peregrine only (I was the person working with them to fix that), but my understanding is that there is at least one other small BLE fix that went in that should help with the issue of not wanting to connect. Or let me phrase it the other way around. With my Teric (exact same BLE hardware/software as the Perdix AI) and v81 I can now download to different computers / phones without having to go through weird Bluetooth maneuvers. Admittedly, that's with the latest development version - 4.9.6 still tries to force pairing...

And YES, BLE is painfully slow. I love the whole "fully sealed dive computer, no moving parts, etc" concept - but the first time you try downloading 100 dives and wait 3+ hours... yeah. Bleeping slow. But for normal day to day use... the Perdix AI and then the Teric have been my main dive computer for quite a while, so on dive trips (back when that was a thing) I'd download after two or three or five dives... and yeah, it takes a few minutes, but it's manageable.

closed 2020-09-10 by dirkhh

ryanm101, 2020-09-10:
@dirkhh 

No Luck. I just built from source 4.9.6-72-gbeedea7ccfc
[image]

```
Detecting bluetoothd version
Using /proc/<pid>/exe
Using /proc/<pid>/cmdline
Detected bluetoothd version 5.54
Bluetoothd: "5.54"
Using BlueZ LE DBus API
Bluez 5 detected.
QLowEnergyControllerPrivateBluezDBus::connectToDevice()
###### "org.bluez.Device1" QMap(("Connected", QVariant(bool, true)))
BTLE_DBUS::connect() failed "org.bluez.Error.Failed" "Software caused connection abort"
Using BlueZ LE DBus API
QLowEnergyControllerPrivateBluezDBus::connectToDevice()
###### "org.bluez.Device1" QMap(("RSSI", QVariant(short, -53)))
###### "org.bluez.Device1" QMap(("RSSI", QVariant(short, -41)))
###### "org.bluez.Device1" QMap(("RSSI", QVariant(short, -52)))
###### "org.bluez.Device1" QMap(("Connected", QVariant(bool, true)))
###### "org.bluez.Device1" QMap(("Connected", QVariant(bool, false)))
Missing CAP_NET_ADMIN permission. Cannot determine whether a found address is of random or public type.
Creating QtBluezDiscoveryManager
Discovered:  "[TV] DenTV" "00:7C:2D:10:0B:62" Num UUIDs 4 total device 0 cached RSSI 0 Num ManufacturerData 2
Discovered:  "45-27-36-AA-A1-C2" "45:27:36:AA:A1:C2" Num UUIDs 1 total device 1 cached RSSI 0 Num ManufacturerData 0
Discovered:  "HM_X1" "80:EA:CA:00:01:1D" Num UUIDs 1 total device 2 cached RSSI 0 Num ManufacturerData 0
Discovered:  "8Bitdo SNES30 GamePad" "E4:17:D8:6B:0B:75" Num UUIDs 2 total device 3 cached RSSI 0 Num ManufacturerData 0
Discovered:  "8Bitdo NES30 Pro" "E4:17:D8:E4:06:7A" Num UUIDs 2 total device 4 cached RSSI 0 Num ManufacturerData 0
Discovered:  "Perdix" "F1:6D:B7:D5:9C:5B" Num UUIDs 3 total device 5 cached RSSI 0 Num ManufacturerData 0
Discovered:  "My speaker F013" "FC:58:FA:B9:2D:7A" Num UUIDs 6 total device 6 cached RSSI 0 Num ManufacturerData 0
Discovered:  "71-BB-7E-66-BF-3E" "71:BB:7E:66:BF:3E" Num UUIDs 1 total device 7 cached RSSI -44 Num ManufacturerData 1
Discovered:  "4A-36-28-7C-0A-11" "4A:36:28:7C:0A:11" Num UUIDs 1 total device 8 cached RSSI -49 Num ManufacturerData 1
Updating RSSI for "71:BB:7E:66:BF:3E" QVariant(short, -58)
Discovered:  "58-F7-E0-7E-55-25" "58:F7:E0:7E:55:25" Num UUIDs 1 total device 9 cached RSSI -48 Num ManufacturerData 1
Discovered:  "67-38-B7-93-12-F7" "67:38:B7:93:12:F7" Num UUIDs 1 total device 10 cached RSSI -80 Num ManufacturerData 0
Updating RSSI for "00:7C:2D:10:0B:62" QVariant(short, -79)
Updating ManufacturerData for "00:7C:2D:10:0B:62"
Almost Duplicate  "00:7C:2D:10:0B:62" "[TV] DenTV" - replacing in place
Updating RSSI for "67:38:B7:93:12:F7" QVariant(short, -81)
Updating ManufacturerData for "67:38:B7:93:12:F7"
Updating RSSI for "45:27:36:AA:A1:C2" QVariant(short, -83)
Updating RSSI for "58:F7:E0:7E:55:25" QVariant(short, -48)
Updating RSSI for "67:38:B7:93:12:F7" QVariant(short, -78)
Updating RSSI for "58:F7:E0:7E:55:25" QVariant(short, -50)
void QBluetoothDeviceDiscoveryAgentPrivate::stop()
Using BlueZ LE DBus API
QLowEnergyControllerPrivateBluezDBus::connectToDevice()
BTLE_DBUS::connect() failed "org.bluez.Error.Failed" "Operation already in progress"
Discovered:  "[TV] DenTV" "00:7C:2D:10:0B:62" Num UUIDs 4 total device 0 cached RSSI 0 Num ManufacturerData 2
Discovered:  "45-27-36-AA-A1-C2" "45:27:36:AA:A1:C2" Num UUIDs 1 total device 1 cached RSSI 0 Num ManufacturerData 0
Discovered:  "HM_X1" "80:EA:CA:00:01:1D" Num UUIDs 1 total device 2 cached RSSI 0 Num ManufacturerData 0
Discovered:  "8Bitdo SNES30 GamePad" "E4:17:D8:6B:0B:75" Num UUIDs 2 total device 3 cached RSSI 0 Num ManufacturerData 0
Discovered:  "8Bitdo NES30 Pro" "E4:17:D8:E4:06:7A" Num UUIDs 2 total device 4 cached RSSI 0 Num ManufacturerData 0
Discovered:  "Perdix" "F1:6D:B7:D5:9C:5B" Num UUIDs 3 total device 5 cached RSSI 0 Num ManufacturerData 0
Discovered:  "My speaker F013" "FC:58:FA:B9:2D:7A" Num UUIDs 6 total device 6 cached RSSI 0 Num ManufacturerData 0
Discovered:  "58-F7-E0-7E-55-25" "58:F7:E0:7E:55:25" Num UUIDs 1 total device 7 cached RSSI -50 Num ManufacturerData 0
Discovered:  "71-BB-7E-66-BF-3E" "71:BB:7E:66:BF:3E" Num UUIDs 1 total device 8 cached RSSI -43 Num ManufacturerData 1
Discovered:  "67-38-B7-93-12-F7" "67:38:B7:93:12:F7" Num UUIDs 1 total device 9 cached RSSI -80 Num ManufacturerData 0
Updating RSSI for "00:7C:2D:10:0B:62" QVariant(short, -79)
Updating ManufacturerData for "00:7C:2D:10:0B:62"
Almost Duplicate  "00:7C:2D:10:0B:62" "[TV] DenTV" - replacing in place
Updating RSSI for "67:38:B7:93:12:F7" QVariant(short, -81)
Updating RSSI for "58:F7:E0:7E:55:25" QVariant(short, -52)
Updating ManufacturerData for "58:F7:E0:7E:55:25"
Updating RSSI for "00:7C:2D:10:0B:62" QVariant(short, -83)
Updating RSSI for "67:38:B7:93:12:F7" QVariant(short, -80)
Updating ManufacturerData for "67:38:B7:93:12:F7"
void QBluetoothDeviceDiscoveryAgentPrivate::stop()
[... 18 lines cut ...]
Updating RSSI for "00:7C:2D:10:0B:62" QVariant(short, -78)
Updating RSSI for "00:7C:2D:10:0B:62" QVariant(short, -97)
Updating RSSI for "58:F7:E0:7E:55:25" QVariant(short, -52)
Updating RSSI for "71:BB:7E:66:BF:3E" QVariant(short, -44)
Updating RSSI for "F1:6D:B7:D5:9C:5B" QVariant(short, -53)
Updating RSSI for "67:38:B7:93:12:F7" QVariant(short, -80)
Updating RSSI for "80:EA:CA:00:01:1D" QVariant(short, -88)
void QBluetoothDeviceDiscoveryAgentPrivate::stop()
Using BlueZ LE DBus API
QLowEnergyControllerPrivateBluezDBus::connectToDevice()
###### "org.bluez.Device1" QMap(("Connected", QVariant(bool, true)))
[... cut ...]
[0.916674] WARNING: Unknown hardware type 7c2d. Assuming Petrel. [in src/shearwater_petrel.c:249 (shearwater_petrel_device_foreach)]
EVENT_DEVINFO gave us a different detected product (model 0 instead of 6), but that one is unknown.
[23.226479] ERROR: Failed to receive the packet. [in src/shearwater_common.c:239 (shearwater_common_slip_read)]
[23.226517] ERROR: Failed to receive the response packet. [in src/shearwater_common.c:358 (shearwater_common_transfer)]
[23.226527] ERROR: Failed to download the dive. [in src/shearwater_petrel.c:380 (shearwater_petrel_device_foreach)]
```

ryanm101, 2020-09-10:
The final fail was when i paired it to the laptop 1st again, it connected but failed to DL dives. 

I'll do a screen recording later and add to this so you can see 1st hand.

dirkhh, 2020-09-10:
Thanks for the logs. I asked Linus to take a look as well - he has written most of our BLE code, maybe he has suggestions what to try

dirkhh, 2020-09-10:
Just heard back from Linus - as I thought, this is a failure of BlueZ to actually talk to your dive computer at all. We never even see a connection from the system library. Typical questions include "Is this builtin Bluetooth or some dongle?" and "does removing the device in BlueZ and re-discovering it still fix the problem"?

ryanm101, 2020-09-10:
Dell XPS running Fedora 32 and built in Bluetooth. 
You can skip between 2:00->3:50 (at 3:50ish i scroll the logs) as that is just waiting as is 4:19->6:32

[BlueZ_issue.zip](https://github.com/subsurface/subsurface/files/5204892/BlueZ_issue.zip)

ryanm101, 2020-09-21:
@dirkhh Is there any more information that I could provide to help with this?

dirkhh, 2020-09-21:
I reached out  to Shearwater - you are getting back an invalid hardware version (0x7c2d) which makes no sense. They weren't able to tell me why you'd get that.
Because of that the libdivecomputer backend returns an unknown 'model = 0' to Subsurface.
Those are the two warnings that you see @3:52 in the video.
The error "software caused connection abort" comes from Bluez and basically says that it lost its connection with the dive computer - so that's way below in the protocol stack, not something that we deal with.
You are using firmware v81, so the current Perdix AI firmware - not sure what else to try here.
I'm basically out of ideas.
You could try downloading with Subsurface-mobile on Android or iOS and see if that gives you better results.

ryanm101, 2020-09-23:
@dirkhh  It works fine if I remove and re-pair, so I can live with it.

If you think of anything else to try let me know, happy to help where i can.