subsurface/subsurface #2921: DiveComputer.eu / tecdiving problems to impot logs

nightzero, 2020-08-20:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I can no longer import divelog to Subsurface. This has always worked before.

HW 3.0
SW 4.1
Subsurface version: 4.9.6
OS: Windows 10.

When clicking "Download" progress bar is not moving. Only say: "Opening rfcomm stream 'device number' "

Then I get errordialog:
"Could not open file 'device number' Tecdiving(DiveComputer.eu)"

And red bar in the bottom say "Unsupported Operation".
This is what I tried:
*Removed device in windows
*Rebooted BT interface in windows.
* Rebooted DC
* In logbook klick a dive and then "Upload"
* Pair the DC in windows again
* Go to Subsurface
* Import -> Import from dive computer
* Choose Tecdive / divecomputer.eu
* Check download mode for Bluetooth
* Search for the DC in the search dialog that appear
* Select the DC in the list and press, "Save"
* Then klick "Download"

At his stage I get the error described above.

As I said this worked before but have now stopped working.

### Operating system:

Windows 10, swedish

### Subsurface version:

4.9.6

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

nightzero, 2020-08-21:
So I downgraded to version 4.9.3 and it works again, i.e. no problem to connect to DC and download logs.

My guess is that the problem is related to rfcomm implementation, and problem was introduced with the QtBluetooth:
https://github.com/subsurface/subsurface/commit/aceb8a547f1ea70b71ab609c6e406928dee74b88

nightzero, 2020-08-23:
@jefdriesen I tried downloading with dctool as per your instruction and it is working just fine.

What conlusion can we draw from this. Is the problim isolated to the subsurface code? Is libdivecomputer-library using Qt bluetooth as well?

jefdriesen, 2020-08-23:
The dctool application uses libdivecomputer's built-in Bluetooth (rfcomm) implementation. Since that works fine, we can rule out problems with the dive computer itself. Subsurface uses libdivecomputer with a custom I/O implementation based on the Qt Bluetooth stack. The main reason for that is because it's available on more platforms (e.g. Mac OS X), and supports not only Bluetooth classic (rfcomm), but also Bluetooth Low Energy (BLE). But the fact that downloading with subsurface fails, while everything else is the same, points me in the direction of the Qt Bluetooth. But I'm not sure what or why is actually going wrong.

That WSAStartup error is certainly something suspicious.

Another thing is that "The connection step took more than expected. Wait another 20 seconds". The bluetooth discovery and connection is no doubt a slow operation. I just tried with my divecomputer.eu unit, and it took about 12 seconds, and I wouldn't be surprised if it can be even longer:
```
[0.001775] Opening the I/O stream (bluetooth, null).
[10.691076] INFO: Discover: address=D88039F337BD, name=DiveComputer
[10.691354] INFO: Open: address=D88039F337BD, port=0
[12.070535] INFO: SDP: channel=6
[12.295215] Opening the device (Tecdiving DiveComputer.eu).
```
Maybe subsurface doesn't wait long enough, and gives up too soon?

dirkhh, 2020-09-20:
@nightzero - a Windows test binary that should fix your problem is available here:
https://subsurface-divelog.org/downloads/test/subsurface-4.9.6-118-g767f0a4f3eb0.exe
Sorry it took so long to get this sorted out.

closed 2020-09-20 by dirkhh

nightzero, 2020-10-12:
Thanks! Did not have the opportunity to try it before, but just did, It works perfect!