subsurface/subsurface #2928: Shearwater Peter dive import broken

pvalsecc, 2020-08-29:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When trying to download dive logs from by shearwater petrel (bluetooth), the download fails. The dive computer shows 
```
BLUETOOTH ERROR:
  Rcv Stream Err
```

### Operating system:
Linux Ubuntu 20.04, english
Shearwater Petrel (first version) with firmware v72

### Subsurface version:
git hash: f5ba42e93
Was working feb/17/2020 , but I cannot tell if it's a firmware upgrade or a subsurface upgrade that broke it since I've done both. 
Compiled myself, as usual.

### Steps to reproduce:
* Start bluetooth on the dive computer
* Import from divecomputer
* Click download
* put 0000 as pin (I was not needing that before!?!?!?)

### Current behavior:
Error

### Expected behavior:
My dives downloaded

### Additional information:
I've asked for libdivecomputer logfile and dumpfile. I've got only a short log:
```
Subsurface: v4.9.6-44-gf5ba42e933ae, built with libdivecomputer v0.7.0-devel-Subsurface-NG (1df0e13cc44041743836065c66928c06a5c035a0)
[0.000005] INFO: Open: transport=16
[0.000031] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[0.000034] INFO: Timeout: value=3000
[0.000036] INFO: Sleep: value=300
[0.300170] INFO: Purge: direction=3
[0.300308] INFO: Write: size=9, data=FF0105002E902000C0
```

### Mentions:

marccollin, 2020-08-29:
I have same issue with a petrel 2

dirkhh, 2020-08-29:
I'd actually like to see the Subsurface logfile. I made some changes when setting up BLE connections, but this is an rfcomm connection and shouldn't be affected.
One thing to try is to tap on the '...' after the Bluetooth checkmark and make sure that you right click on your Petrel and pair it before trying to download from it...

marccollin, 2020-08-29:
log gave me: Subsurface: v4.9.6, built with libdivecomputer v0.7.0-devel-Subsurface-NG (48e46cf7775452740f67ba72d9c5a0960ebf3718)

on windows i was able to get data, but get: error parsing the header...

dirkhh, 2020-08-29:
Have you tried pairing as I suggested?
The log above seems to indicate that we fail starting the conversation with the dive computer - and latest master no longer forces you to pair (as many dive computers don't actually need that step).
I just did a successful download (on Mac) with latest master from my Petrel 2 after pairing it in the Remote Bluetooth device selection dialog, but there are actually three subtly different version of the Petrel 2 out in the wild...

marccollin, 2020-08-29:
yes is paired 
[https://pasteboard.co/JoF15Hj.jpg](url)

dirkhh, 2020-08-29:
I clearly haven't updated my firmware in a while... mine is, err, a bit older.
What I do notice is that yours is doing a BT download, while mine was doing a BLE download. Let me try to force a BT download as well to see if that's at the core of the problem...

marccollin, 2020-08-29:
tried with 4.8.4 release no problem...

retried with the last version.... i unpaired it, paired.. we can see communication between computer and petrel...... but seem like to download all dive..... there is no new dive

[https://pasteboard.co/JoFGuco.jpg](https://pasteboard.co/JoFGuco.jpg)

for every dive imported, there is an error

dirkhh, 2020-09-09:
I have been able to reproduce this with the latest master. We appear to have broken rfcomm downloads from older Shearwater non-BLE dive computers. Added to my long list of "things to look at"...

BTW: Shearwater released new firmware for most of their dive computers which actually fixes a couple of Bluetooth issues. I don't think that it fixes THIS one, but it might still be worth updating to.

marccollin, 2020-09-17:
i have lastest firmware, i was able to get data under windows only, not with linux

dirkhh, 2020-09-17:
I'm super curious about the "worked on Windows" - given that Linus and I think that the current implementation of rfcomm is actually truly broken. I need to try on my Windows box and look if there's something fundamentally different we are doing there.

pvalsecc, 2020-09-19:
I've tried with a 4.8.6 and didn't work either using the Bluetooth download mode. Didn't work either.
But unchecking the bluetooth download mode and setting up the serial port manually (`rfcomm connect`), I can download my dives without any problem.

Subsurface-divelog, 2020-09-19:
rfcomm downloads as with the older Petrel dive computers appear completely broken right now.

Linus and I are trying to fix this, but so far we are stuck. Hang in there, I'm sure we can figure it out, but it may take a couple of days.

dirkhh, 2020-09-20:
@pvalsecc I have test binaries available in the subsurface-daily repo for Ubuntu. Can you look for Subsurface-4.9.6-118 in there?
I believe with that downloads from the Petrel should work again. I'd appreciate any feedback you might have.

A Windows binary for the same version is available here: https://subsurface-divelog.org/downloads/test/subsurface-4.9.6-118-g767f0a4f3eb0.exe
This should solve the parse errors that you saw @marccollin  - again, confirmation would be appreciated as I want to release Subsurface 4.9.7 with these (and other) fixes. Thanks

marccollin, 2020-09-20:
not tried linux version, don't have ubuntu, but under windows i get error about not able to open shearwater petrel 2 when opening rfcomm adresss ...

Subsurface-divelog, 2020-09-20:
Did you unpair and re-pair?
Windows is weird that way.
I tried three times with my Perdix 2 and it reliably worked (Dell laptop, built-in Bluetooth, win 10 fully up to date)

And of course all the other usual Windows suggestions, like reboot, dance on one foot, chanting in German, etc...

Subsurface-divelog, 2020-09-20:
Forgot to ask, which Linux do you have? We have binaries for Fedora and opensuse as well...

marccollin, 2020-09-20:
opensuse

dirkhh, 2020-09-21:
For openSUSE, test builds can be found here: https://software.opensuse.org/download.html?project=home:Subsurface-Divelog&package=subsurfacedaily

closed 2020-09-21 by dirkhh

pvalsecc, 2020-09-21:
> @pvalsecc I have test binaries available in the subsurface-daily repo for Ubuntu. Can you look for Subsurface-4.9.6-118 in there?
I believe with that downloads from the Petrel should work again. I'd appreciate any feedback you might have.

I've tested with #c33dd3808 with the Bluetooth download mode on Linux Ubuntu and it works. Thanks a lot @dirkhh for the fix!