subsurface/subsurface #2929: Feature Request: Autosave

mwerle, 2020-09-02:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

I've several times now had Subsurface crash on me while performing large scale edits (most recently a major renaming of dive sites to a common format) on my Cloud Save.

Of course, all unsaved edits are lost.

Most modern software includes autosaved recovery data these days.

It would be very nice if Subsurface could save its state to a temporary file while performing edits.  Ideal would be some dump of the current Subsurface state which includes undo/redo data, so when restarting Subsurface, it would resume at almost the same state that it was in when it crashed.

### Operating system:

Various; Windows 10 and Debian GNU/Linux most commonly

### Subsurface version:

Various; most recently 4.9.6 (official release)

atdotde, 2020-09-02:
We could do this in an extra branch...

dirkhh, 2020-09-02:
Far more complex than it may seem. Maintaining state (including undo state) is entirely non-trivial.
We could adapt the desktop to a pattern more similar to what mobile does, which is to sync with local git after most actions.
I absolutely LOATH that behavior and would insist on it being off by default, but it seems the most likely way to achieve what @mwerle is asking for.

Personally, I prefer an approach that puts the responsibility on the user. Save if you want things saved. But don't force the user to save things if that's not what they want.

atdotde, 2020-09-02:
Maybe I am naive (and I still haven't tried to understand the whole undo-business), but what kind of state needs to be managed?  

When using cloud mode, I would save to a local branch [email@example.com-BACKUP] every 30s say but without removing the "there are changes to be saved"-state. When the user explicitly saves (as it is now), it does the save and deletes the backup branch. Upon opening cloud storage, if there is a backup branch offer to merge it via the UI. What could possibly go wrong?

dirkhh, 2020-09-02:
Timed save seems very challenging as you would have to lock the internal data structures so they aren't changed while you are saving - we don't support saving in a separate thread like that.
You could save to your branch after every transaction, but that's different from what was suggested here. It doesn't give you that undo capability, unless you have a UI that allows you to walk backward and forward through your git history. And I don't see anyone volunteering to create that UI, TBH
So what could possibly go wrong? Complete data loss for the user if this ends up creating lost data in their local repo that can't be replayed from the cloud server - and I assume you are not proposing to sync those constant backups to the server.
Could this be solved? Sure. Are there 20 other things that I believe are more important to fix? You bet.

Subsurface at this point has maybe 3 active developers. That's being very generous. It has three people who occasionally find time to do things, and another half dozen who very occasionally, in very narrow areas maybe do a tiny bit of work. That's very different from just a couple of years ago.

atdotde, 2020-09-02:
You know this much better than I do. But I fail to see how the whole undo mechanism is touched at all.

I agree that doing a save in the background can create all kinds of nasty race conditions and should be avoided. So better do it with the transactions as you suggest. Of course, every keystroke is far too often, but what about every transaction, we check of the last backup was saved more than 30 seconds ago? I admit this would be annoying if it stops the UI for more than a moment, in that case I would be strongly against this.

OK, now I reread the original proposal. And yes, saving undo/redo data to the backup is too difficult. But simply offer the current state (i.e. 30s before) before a crash (without the possibility to undo the last keystrokes) upon restart might still be better than nothing and not too hard to do (assuming that the actual save of the backup is almost instantaneous).

dirkhh, 2020-09-02:
git save is not at all instantaneous. We have improved this in the past, but there are several operations that cause us to invalidate most or even all of the cache and depending on the size of the dive list and the system you are on, this can take a few seconds.
But the real problem that you are missing is this: let's say things crashed and you have some changes that are now in that magic branch. You want the first six of them, but not the last five.
How do you propose to give that control to the user? You need a UI that can step through the git commits and that can visualize to the user in a user friendly way what has actually happened.
We have a very rough approximation of what was done in git commit messages - which would not work for this at all because it is session based, not "change since last save" based. I use them when helping people sort our data loss via changes to the backend git storage. Looking at these plus the git diffstat or sometimes the git diff makes it possible for me to guess what actually happened. That is not something you could ever present to a user.

mwerle, 2020-09-03:
> Personally, I prefer an approach that puts the responsibility on the user. Save if you want things saved. But don't force the user to save things if that's not what they want.

Users will _not_ constantly save after every edit.  Even in my case where I was getting semi-regular crashes I'd only save after a "batch" of edits, which often resulted in losing a bunch.

There's a reason a lot of software has auto-save. Whether the auto-save is "permanent" or only "temporary" for recovery purposes is up to each software philosophy, and yes, I fully agree that the permanent save should be under user control.

> git save is not at all instantaneous. We have improved this in the past, but there are several operations that cause us to invalidate most or even all of the cache and depending on the size of the dive list and the system you are on, this can take a few seconds.

Not sure creating and maintaining local git branches is a valid approach; yes, Subsurface uses git for cloud, but (AFAIK) not for purely local files.

But saving will always take some time, regardless of git or xml or some binary memory dump.

So, any possibility of duplicating the current data model in-memory, then saving that "snapshot"? A memory copy should be fairly fast (at the cost of memory, of course).

Alternatively, any possibility of adding signal handlers, so that if a program crash is detected, the current data can be saved out? Plenty of time in this case..

> But the real problem that you are missing is this: let's say things crashed and you have some changes that are now in that magic branch. You want the first six of them, but not the last five.

I think this scenario is almost entirely unlikely as compared to recovering edits done in the last X minutes.

As for saving the undo/redo stack, that was purely icing on the cake. I anticipated this to be almost impossible unless it was built into the software design from the start.

The core feature request is just having some sort of temporary save in the event that the program crashes (in an ideal world it would never crash, but...) and on startup the user is given the option to resume from the recovery save (if one exists) or not. 

Ultimately it's just a "request" though; I appreciate there's very little developer time available.

dirkhh, 2020-09-03:
There are lots of programs that have auto-save. I think I have turned this off wherever one can turn that off. But that's just me. Old and stubborn. I constantly get reminded by the Subsurface and especially Subsurface-mobile user community that I am just an old idiot, so... 🤷🏼‍♂️
You can indeed use XML files, though I don't understand why anyone would - they don't give you the infinitely recoverable integrated backup of git storage. I save after every single edit that isn't an experiment. So any change that I want "kept", I immediately save. A typical editing session of my dive log after a day's worth of diving often results in 20 or more commits to my local repo. I don't think I have used saving to XML files outside of the desire to export something to share with someone else in many, many years. If we were to implement a mechanism that saves after every change, then the only sane way to do that would be to a local git repo. 
I am not aware that Qt offers a way to save state as a snapshot of memory, and the idea of a binary memory dump of our internal data structures in order to be able to auto-save seems like a really bad idea.
As for signal handlers that save once you crash. Umm... a crash almost always indicates that your internal data structures are corrupted. The last thing you'd want to do is write data to disk in that state. The only times I see actual irreversible damage to git storage is when a user's system crashes while they are writing to disk...

If I look at your "core feature request" as expressed above,

> just having some sort of temporary save in the event that the program crashes (in an ideal world it would never crash, but...) and on startup the user is given the option to resume from the recovery save (if one exists) or not.

I think this is something that could be done using a similar pattern to what we do on mobile. The challenge here would be how do you detect that you have newer data in that backup data store. If the user uses a default data file, you could obviously try to compare time of last change (trivial with local git storage, doable with a local XML file). But what do you do if the user was editing a different file... hmm... I guess we could include the data store URL as part of the meta data in the backup... I'll need to think about the architectural implications a little more.

mwerle, 2020-09-04:
> As for signal handlers that save once you crash. Umm... a crash almost always indicates that your internal data structures are corrupted.
I have to bow to your experience here; I was just thinking of the root cause of my last bout of crashes which were due to a logic error in the code.  I do agree that a corrupted data state would be very difficult if not impossible to save (and if you did, would it just cause the program to crash again on recovery?)

> I'll need to think about the architectural implications a little more.
Thank you for considering; it's certainly not a high-priority issue though; just (imho) "nice to have" in the event of a program crash. Of course, the priority should be to never have a crash...

bstoeger, 2020-09-09:
Writing automatically to an XML or other single file seems more practical to me than a git repository. Much simpler - and in this case you definitely *want* simple! It's a well understood feature: lock main thread - duplicate data in memory - write out in writer thread with a well known naming scheme. It's time to create a `struct DiveLog`, which collects all the data that is saved to the log (dive, trips, sites, filter, ...). Not only for this, but to finally get control over all the horrible global state. 

If there is interest, I can try to work on the core data structures needed for this.

bstoeger, 2020-09-09:
> I think this is something that could be done using a similar pattern to what we do on mobile.

Please no - the saving behavior of mobile drives me mad.

> The challenge here would be how do you detect that you have newer data in that backup data store. If the user uses a default data file, you could obviously try to compare time of last change (trivial with local git storage, doable with a local XML file)

You're probably overthinking this. Do as numerous other applications ranging trivial text-editors to highly complex multimedia editors do: On program start look for stale backups and present user with "do you want to restore auto-save from DATE" dialog. After a successful save, delete previous auto-backup. Why reinvent the wheel...

dirkhh, 2020-09-09:
The saving behavior on mobile is frustrating. It's the result of us responding to user complaints about lost data since somehow people assume different things to be true on mobile compared to desktop.
I don't agree with those people... but that's how we get here.

It's funny that every time I push back against a user request one of the developers says "well, actually, we could do that". And then we do it. And then we hate it. I can literally go on for half an hour about all the features that we have added because one or three or some other small number of users thought "this" would be great to have. And we have had all that crud forever.

So - sure. We could add a timed autosave to XML to the desktop version. Here's my prediction
(a) we will find massive race conditions and data consistency issues if we try to do this in the background
(b) the synchronous part of this will be annoying for people on slower machines or with really large dive lists
(c) the number of users who ever actually benefit from having this vs the number who never ever see that it exists or who every time they start press 'cancel' 'no' 'GO THE HECK AWAY' because they intentionally Ctrl-C'ed out of the last edit... yeah, I bet you a Euro that the second group is an order of magnitude bigger than the first.

I'll say what I usually say, what I way too often say. If someone wants this enough to send a PR, I'll look at it and likely accept it. It is absolutely not something that I would want to work on. But then, I rarely work on anything anymore besides keeping the infrastructure alive and building binaries and providing end user support for the cloud storage, so... whatever.

bstoeger, 2020-09-10:
> It's funny that every time I push back against a user request one of the developers says "well, actually, we could do that". And then we do it. And then we hate it. 

Sounds like confirmation bias - you probably don't remember the cases where we don't "hate" it. 

> we will find massive race conditions and data consistency issues if we try to do this in the background

I doubt it. First step: remove *all* global state from the parsers. This should be done anyway. Then there can't be race conditions.

> the synchronous part of this will be annoying for people on slower machines or with really large dive lists

Don't turn it on, then.

> the number of users who ever actually benefit from having this vs the number who never ever see that it exists or who every time they start press 'cancel' 'no' 'GO THE HECK AWAY' because they [...]

I don't understand what you mean with the last part, but: don't turn it on then.

dirkhh, 2020-09-10:
> Sounds like confirmation bias - you probably don't remember the cases where we don't "hate" it.

I'm sure you are right.

> > we will find massive race conditions and data consistency issues if we try to do this in the background
> 
> I doubt it. First step: remove _all_ global state from the parsers. This should be done anyway. Then there can't be race conditions.

That's gnarly Linus C code... just saying.

> I don't understand what you mean with the last part, but: don't turn it on then.

Look - if you want to write this, be my guest. As you know from lots of experience, I typically don't stop new ideas. I occasionally try to temper the enthusiasm for adding more and more and more, but in the end, I rarely ever turn down PRs.

mikeller, 2026-10-18:
Hi @mwerle,

This remains a genuinely architectural question (data-structure refactoring, threading/locking, crash-safety semantics) rather than a quick feature bolt-on, per the detailed trade-off discussion already captured in this thread. We're not able to just pick one of the discussed approaches (periodic git snapshots, bstoeger's unified DiveLog + XML-snapshot idea, or the simpler "offer to restore a stale autosave on startup" pattern) and implement it without a deliberate design decision first.

Is this something a maintainer wants to formally scope as a dedicated design effort? If so, we'd want to work from the trade-offs already discussed here rather than starting over. We're not proposing a quick fix, this would need proper design time before any implementation.