subsurface/subsurface #2936: Discussion: Add a divesite location string

mwerle, 2020-09-03:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [X] Discussion request

### Issue long description:

Currently, dive sites have a "name", a "location" (which are GPS coordinates), and potentially some reverse-geocoded tags.

##### Issue 1 - choosing dive sites when entering dive information
The issue is that if a dive site is literally given a name (eg, "Manta Point"), then it becomes very difficult when entering dives to choose the correct dive site for a dive (eg, there may be many "Manta Point" divesites in a users' file, each in a different geographical location).

Having the pop-up include the reverse-geocoded tags (if they are available) would alleviate this issue but not remove it as the user has no control over the tags.  There could be several sites with the same name falling into the same set of tags, so the user would need a way to add further disambiguation information without reverting to including location information in the site name itself which is what I've been doing so far:

eg:
* Australia, QLD, Stradbroke  Island, Manta Point
* Phillipines, Malapascua, Manta Point
etc

This approach doesn't "play nicely" when exporting dive (sites) to other services. It also looks "messy" when viewing the dive sites on the map.

##### Issue 2 - search for all dives in a particular location
Currently when filtering, only the dive site name is used, not the geolocation tags.

Even if geolocation tags would be searched, they don't necessarily "play nicely".  An example is the Izu Peninsula south of Tokyo in Japan. I'd like to list all dives / dive sites in Izu, but the reverse geolocation data places the sites into various different prefectures.  My personal naming strategy for dive sites here is "Japan, Izu, <town>, <site>" - but see point above about this not playing nicely.

##### Issue 3 - playing nicely with external dive site / dive log managers
Most external dive site / dive log managers split site information into a name and a location (where location is a human-readable string, not a set of GPS coordinates). Some leave location as a free-form string, others split it into geographic regions (eg, country, state, city, etc).

Currently, Subsurface only exports the "name" part of a divesite; any reverse-geolookup data is not sent.  This means that the primary export (to divelogs.de) requires a lot of manual work to re-enter the dive site location for each dive. And even if geolocation data would be sent, see my comment above about how this is insufficient.

#### Suggested solution
1 - add a "dive site location" field which is a human-readable string, not GPS coordinates.

This could possibly be filled in automatically by the reverse geolocation lookup, but only if it is empty, and it must allow for editing afterwards.

With this:
- dive-site suggestion popup can show both fields to disambiguate between sites with the same/similar name (show coordinates and/or reverse-geocode lookup data if site-location is empty)
- filtering/searching can operate on both of these fields.
- the user can go to as much granularity, or renaming, of the location as they require.
- this approach "plays nicely" with most external dive log / dive site systems.

dirkhh, 2020-09-03:
we have gone through an insanely painful process to get where we are. If I was able to go back in time the decision to add the concept of a dive site is one that I would definitely want to revert. Different people have very, very different visions how to use these dive sites. And the UI challenges with all of these visions are complex.
So... personally I really don't care... no, that's not at all strong enough... I couldn't care less what other apps / logs do.
I have no interest of another round of "but dive sites should be like...." discussions.
From experience I can say that changes like what you propose would be extremely disruptive, would need massive UI updates in both desktop and mobile, and would certainly require non-backwards-compatible changes in both XML and git file format.

mwerle, 2020-09-04:
Sorry, didn't realise this had been discussed "to death".

Actually while digging through the code for something else I noticed the dive site "taxonomy" data already has provision for being user-editable.  Would you be amenable to someone adding UI elements for leveraging and editing those?

Feel free to close.

bstoeger, 2020-09-09:
Note that IIRC @dirkhh was also skeptical of the dive site tab - and yet here it is. So if you come up with something that people like, I wouldn't write it off.

Personally, I simple use "Bunaken, Ron's point" as name and that is sufficient for my purposes.

Editing taxonomy might be quit a challange UI-wise, since you would have to add a table of sorts. Certainly doable,  though.

mwerle, 2020-09-10:
Here's a discussion about this on ScubaBoard which specifically mentioned Subsurface : https://www.scubaboard.com/community/threads/dive-site-naming-protocol.599666/

I'm currently playing with an idea.  Instead of adding extra fields to split the dive site name into components, which requires adding new data structures, save-file changes, and UI, simply add logic which can _optionally_ perform the splitting depending on formatting in the site name.

I've got it working on the map already, but could be extended to the export function. And would need a couple of preferences  fields in the UI.

Basic premise is to allow users to use separator characters in their dive site name which the software then uses to split the field in certain circumstances.

For example:

* Japan, Izu Peninsula, Atami; Chinsen-Aft
* <Location,fields>;<Site,fields>

Or

* Cod Hole, Ribboon Reef/Great Barrier Reef,Cairns,QLD,Australia
* Coral Garden, Ribboon Reef/Great Barrier Reef, Cairns, QLD, Australia
* <Site,fields>;<Location,fields>

Configuration options (need to come up with good names):

* Field Separator : ","
* Site Separator : ";" 
* Order : Location, Site | Site, Location

This would be completely optional with zero impact on existing users and data and would give complete freedom in what bits of the name belong to the "Location" and which parts to the "Site".   This _should_ (haven't investigated) also work nicely with the divelogs.de export (our primary one?) which only distinguishes between site location and site name.

Thoughts?

dirkhh, 2020-09-10:
Can you help me understand how that would be presented to the user and how the user would use it?
Obviously a user could already use such structured site names, so what does it **mean** that we have the separator defined and the order defined in the larger context?

mwerle, 2020-09-10:
How would a user use it?

They would specify the separator character(s) in the preferences window and that's it.  I notice some existing settings have a large blurb of text near them explaining the setting and how it is used so we could leverage the same approach for people who don't read manuals.

If they already have a naming scheme which uses separator characters, it all immediately works, otherwise they'd need to go through their sites and rename them appropriately.

---

I'm not sure I understand by "what does it **mean**"?  It doesn't "mean" anything; it's just a way to split the existing site name into a location and site for display (and export?) purposes.

For example, currently if I have a site name including the full taxonomy (eg "Cod Hole, Ribboon Reef/Great Barrier Reef,Cairns,QLD,Australia"), that entire string is displayed on the map.  Having several sites close to each other makes the map view messy at best and unreadable at worst.

Splitting a (much shorter) part of the full name out for display purposes makes the map readable.  I don't see a need to display the full taxonomy of a site name in the map since the map already provides the relevant context.

Does this answer your question? Otherwise, could you please reword it as I may have misunderstood.

dirkhh, 2020-09-10:
Oh cool. That makes sense. And is easy.
I like it

mturkia, 2020-09-10:
Our import / export functionality already use this to some extent. E.g.
divelogs.de export looks for / character and splits the location string to
site and location fields. Similarly many of the imports construct a
location from multiple fields separating the different parts with a slash
character.

mwerle, 2020-09-11:
> Our import / export functionality already use this to some extent. E.g. divelogs.de export looks for / character and splits the location string to site and location fields. Similarly many of the imports construct a location from multiple fields separating the different parts with a slash character.

Thanks, so at the very least we should default to using "/" and whatever order is used.

I just found the relevant code, and... well, words fail me.  To export, you build some sort of custom intermediate XML file in memory (not using an actual XML object builder, nor shared with any of the other exporters), import it into an XML object, run that through an XSLT converter, and then finally converting it BACK to a string and saving it.  Seems incredibly convoluted instead of just generating the output file directly.  I could sortof understand this if the intermediate XML file was a shared internal format which all the other exporters would use to generate their output.

Anyway, I have no idea how I can "parameterise" the XSLT (along with QT, I also don't speak XML) to use a different separator character or a different order, or even if that is possible.  Something that, in theory at least, seemed really simple and easy to implement doesn't appear to be. :(

mwerle, 2020-09-11:
So the "simplest" solution is to document what we have (turns out it's "location/site" and hard-code that into the map display widget to look nicer. 

In future we can look at making it a preference setting (my personal preference is  "location, site") if there's a relatively easy way of parameterising the XSLT(s) for the export(s).

bstoeger, 2020-09-11:
Whatever you do - please make it optional. I don't use divelogs.de and therefore do not use '/' as a location-site separator.

mturkia, 2020-09-11:
Parameters for the XSLT are quite simple, so that should pose no problem.

The XML you call intermediate XML is Subsurface's own format that used to be the only saving option back in the day. Imports and exports were naturally built on top of the format we had at the time.

closed 2022-08-22 by dirkhh