subsurface/subsurface #2937: No way to edit or sort cylinder list

sveinmagnus, 2020-09-04:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:
There is currently no way to sort or edit the list of available dive cylinder types, except by filtering unused cylinder types.
When a newly added cylinder is wrong due to a typo it would be nice to be able to correct or delete it.

### Subsurface version:
4.9.6 official release

bstoeger, 2020-09-09:
Of course you can edit and delete cylinders. It is true though that you can't *reorder* cylinders, if that is what you mean..?

sveinmagnus, 2020-09-09:
Well editing or deleting would help a lot, but I cannot figure out how to do this. There are no apparent places to edit the cylinder list.

bstoeger, 2020-09-09:
Perhaps we are talking at cross-purposes here? In the equipment tab simply click on cylinder type, etc. as in the attached screenshot:

[image]

sveinmagnus, 2020-09-09:
Yes you can edit the cylinder type for a dive, and you can also add new cylinder types there. But you cannot edit N existing cylinder type, for instance if you make a mistake and add a AL100 and set it to be 50 liters, then you cannot correct or remove thiis cylinder type after, there will always be the erroneous AL100 in the list.

dirkhh, 2020-09-09:
>  But you cannot edit N existing cylinder type, for instance if you make a mistake and add a AL100 and set it to be 50 liters, then you cannot correct or remove thiis cylinder type after, there will always be the erroneous AL100 in the list.

So, you are partially right, partially wrong.
The list of cylinders that you see is a combination of two sets of data:
- a hardcoded list of cylinders that are always available, defined here https://github.com/subsurface/subsurface/blob/bee2dea7ccfcabff512b0a2d487dcab768fab8d9/core/equipment.c#L231
- all cylinders that are defined in your dive list

Because of the way that we maintain that list in memory, you may have to restart Subsurface before a cylinder you have removed from your dive list is also removed from the list of available cylinders.

So, in your example, if you add a 'myAL100' cylinder with 50 liters of dry volume, once you remove this cylinder from being referenced in your dive list, it will go away again.

This may not be the best possible way to deal with cylinders, but Subsurface is already way, WAY, too configurable. Making the cylinders list another configurable option will only cause more pain for very limited gain.

bstoeger, 2020-09-09:
> This may not be the best possible way to deal with cylinders, but Subsurface is already way, WAY, too configurable. Making the cylinders list another configurable option will only cause more pain for very limited gain.

Yes, this is somewhat hairy - but what about removing unreferenced non-hardcoded cylinders after a cylinder is edited? I think the common solution would be to add a reference count and init all hard-coded cylinders to a reference count of 1. Of course, in the light of the undo commands, this is all a bit tricky - everytime we swap dives we'd have to update the refcount. This is where reference-counting smart pointers are interesting.

dirkhh, 2020-09-09:
I agree that this is of course a solvable problem. The question to me is this: in the list of dozens and dozens of ideas that we have and that have been brought to us by users... how high up is this? Today, if you delete the last reference to a specific cylinder, save the data, and restart Subsurface, that cylinder is gone. That seems like a reasonably useful compromise. 

Could we do more? Could we make this all fully configurable and dynamic and perfect? Sure. Should we? I have a few other things I'd like to see fixed at a higher priority.

sveinmagnus, 2020-09-09:
Thanks dirkhh! 
I had not noticed that unused cylinders got removed on a restart, but that solves the immediate issue for me. I would still love to have the cylinder list be sorted so my custom cylinders appear in the correct order among the hardcoded ones on the list, but that is less important than getting rid of erroneous ones. 

Regarding whether Subsurface is too configurable already I just want to point out that having a configuration page to edit equipment lists such as cylinders and suits are available in other "simpler" products such as Suunto Dive Manager, but I agree that having the lists be managed "automatically" depending on use is pretty good as well.

Other than leaving this open for implementing list sorting I consider this issue resolved.

bstoeger, 2020-09-09:
Perhaps simple generate the list on demand? We'd have to test this with the huge log - if it is reasonably fast, this could make the code distinctly simpler.

dirkhh, 2020-09-09:
Making this list dynamic / on demand should be fairly quick. I like the idea of having it sorted which could be easily done that way as well. If we constrain the proposal to that, then I agree that this sounds reasonably straight forward and useful.

As for performance - you'd have to walk the dive list (something like <20k items) and compare every cylinder (usually zero to maybe 4 or 5) to the existing list of cylinders (hopefully <100). That will likely need some careful coding to make it efficient, but it doesn't sound horribly complex.

mturkia, 2020-09-10:
For me the current implementation has always been more than enough. I just type the start of the cylinder description, and get a short list of matching cylinders. (You need to type into the description field, without opening up the drop-down list.) Then it is easy to pick the one you are looking for.

However, it would be slightly better if the cylinders were sorted, and it would be slightly better if the matching would not be for the start of the string. I.e. currently I type "al" and get the cylinders starting with AL and ALU, but typing "10" gives me nothing, when it could produce a list of 10l, AL100, HP100, etc.

bstoeger, 2020-09-10:
> As for performance - you'd have to walk the dive list (something like <20k items) and compare every cylinder (usually zero to maybe 4 or 5) to the existing list of cylinders (hopefully <100). That will likely need some careful coding to make it efficient, but it doesn't sound horribly complex.

Hashes are your friend. There is one problem though: we somewhere have to save the "default size", so we can't do away with the TankInfo-list.