subsurface/subsurface #2938: Changing weight type resets weight amount

sveinmagnus, 2020-09-04:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When changing the weight type for a dive the weight amount field is reset to either 0 or to last used weight amount of the given weight type. 

### Operating system:
Windows 7 English

### Subsurface version:
Subsurface 4.9.6 official

### Steps to reproduce:
Go to a dive log entry with a unknown weight type and a specified weight amount set
Change the weight type to any other type. 
the specified amount is changed to 0 for unused weight types, or to the previously used amount for the specified weight type.

### Current behavior:
weight amount is set along with weight type

### Expected behavior:
weight type and amount should be independently editable

### Additional information:

### Mentions:

bstoeger, 2020-09-09:
It's not a bug - it's a (mis)feature. Indeed, when changing the weight type, then the previous weight used for that type is set.

I think the idea was that when you edited multiple dives of a trip, you generally use the same weights. However, with the multi-edit functionality this feature might be moot. It's certainly up for discussion.

bstoeger, 2020-09-09:
I assigned a few developers, whom might have an opinion on this.

glance-, 2020-09-09:
I'm used to it, and it's bin like that for like forever, and I kinda like that feature.

Both fields are independently editable, it just autofills the last used weight.

I'm open to other suggestions, but just removing it feels like a loss of functionallity.

dirkhh, 2020-09-09:
I agree with Anton - this has been the design since forever.
Simply in your editing process, first set the weight type, then set the weight (left to right) and you'll never have a problem with this. And if you are like most people and typically dive with the same amount of weight, then this is actually quite an intuitive and convenient design.

sveinmagnus, 2020-09-09:
I understand that this feature might be practical when logging new dives, but when editing large amounts of existing imported dives it becomes problematic. For most imported dives the weight amount is already set as this is supported in most software, however the weight type is usually not. 
When mass-editing such imported dives to set the missing weight type this causes the existing weight amounts to be reset to the previously used weight amount for the chosen weight type, which causes the loss of the originally stored weight amount. This is especially problematic for travel divers like me who have a lot of dives using rented equipment, as well as dives with my own equipment but in highly varying locations so that the weight amounts vary a lot, even for the same weight type.

I would suggest as a good compromise to keep this functionality as is when editing new dives where the weight amount has not been set previously to keep it stable for long-time users, but to make a minor modification so that the weight amount is left alone if it has already been defined, whether manually or by import, as this would make more sense for historical edits.

dirkhh, 2020-09-09:
What I think you are hearing here is that the current developers in general are happy with the design that we have in place.

I am certainly happy to entertain a pull request that adds this capability, but I don't think one of us will be working on this.

sveinmagnus, 2020-09-12:
Thanks a lot @bstoeger!
When will this be in an official release?

bstoeger, 2020-09-12:
I don't know if it will be accepted at all. You can help by testing the feature.

If you don't know how to build a pull request, you use the automatic build. E.g.: https://github.com/subsurface/subsurface/actions/runs/251174557

closed 2020-09-12 by dirkhh

atdotde, 2020-09-13:
I have no opinion as I never log weights myself (even when experimenting with weights, my weights are "big v-weight" and "small v-weight" since I made those myself and I don't even know their weight. The currently proposed solution (only overwrite 0 weights) sounds reasonable to me.

commit 9c90e140 (2020-09-12): desktop: on weight type change, don't overwrite weight if already set

When importing from other software, it happens that weights are imported
without their type. When the user changes the type, the imported weight
is overwritten, which is not exactly a friendly behavior.

On the other hand, when changing the type after creation of a weight
entry, it is preferrable to set a default weight. This is convenient
for people who commonly use the same weight.

As a compromise, set the default weight only if it was unset. We
recognize this by a weight value of 0 g.

Fixes #2938
commit ed847b58 (2020-09-12): desktop: on weight type change, don't overwrite weight if already set

When importing from other software, it happens that weights are imported
without their type. When the user changes the type, the imported weight
is overwritten, which is not exactly a friendly behavior.

On the other hand, when changing the type after creation of a weight
entry, it is preferrable to set a default weight. This is convenient
for people who commonly use the same weight.

As a compromise, set the default weight only if it was unset. We
recognize this by a weight value of 0 g.

Fixes #2938
PR #2946 (2020-09-12): desktop: on weight type change, don't overwrite weight if already set
When importing from other software, it happens that weights are imported
without their type. When the user changes the type, the imported weight
is overwritten, which is not exactly a friendly behavior.

On the other hand, when changing the type after creation of a weight
entry, it is preferrable to set a default weight. This is convenient
for people who commonly use the same weight.

As a compromise, set the default weight only if it was unset. We
recognize this by a weight value of 0 g.

Fixes #2938

### Describe the pull  [... cut]
commit 72312bec (2020-09-12): desktop: on weight type change, don't overwrite weight if already set

When importing from other software, it happens that weights are imported
without their type. When the user changes the type, the imported weight
is overwritten, which is not exactly a friendly behavior.

On the other hand, when changing the type after creation of a weight
entry, it is preferrable to set a default weight. This is convenient
for people who commonly use the same weight.

As a compromise, set the default weight only if it was unset. We
recognize this by a weight value of 0 g.

Fixes #2938