subsurface/subsurface #2939: Sync to cloud should save to file before syncing

sveinmagnus, 2020-09-04:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
when syncing a dive log to cloud, it should also be saved in the open local file

### Operating system:

### Subsurface version:
Subsurface 4.9.6 official

### Current behavior:
When syncing a dive log to cloud, the log is not saved to disk at the same time before beginning the sync. If the sync to cloud causes the application to crash, any edits not saved to file previously will be lost.

### Expected behavior:
When a sync is initiated manually, the log should be saved to the open file that is being edited before beginning the sync to ensure no data is lost if the sync causes the application to crash or the sync is somehow incomplete.

### Additional information:

### Mentions:

dirkhh, 2020-09-04:
Subsurface and Subsurface-mobile both always save to local disk first, and then sync that local repo back to the cloud.

sveinmagnus, 2020-09-05:
If that were the case I would not have had data loss when SubSurface crashed upon sync, when I did a "manual cloud upload" after having added locations to a lot of dives. The application crashed, and when reopening the changes were not stored in the local file, and they did not become available when I tried resyncing with the cloud either.

sveinmagnus, 2020-09-05:
@dirkhh I have updated the issue to show it as a bug to the functionality you describe, as it does not always save to disk first.

dirkhh, 2020-09-05:
Your problem is that the app crashes before ever saving anything at all.
Now what would be interesting is to figure out why the app crashes - because that's not good.
We have just this week found and fixed a bug with taxonomies / reverse geo lookups that could cause crashes.
Can you reliably reproduce the behavior? Can you explain exactly which steps are creating the crash (because, obviously, this doesn't happen for me when I add locations to dives).

sveinmagnus, 2020-09-05:
Sorry I can't recreate it, after reopening and retagging the dives I saved it manually to file first, and then the sync worked as expected. Also it didn't crash immediately, but went through at least half of the sync-steps over several seconds (maybe 5?) before crashing, so I find it unlikely that a save to disk wouldn't have happened already at that point.

sveinmagnus, 2020-09-05:
Actually I just recreated it by accident. I had updated star-ratings on my dives, and upon doing a regular save it launched a cloud sync which crashed the application. Upon restart no star ratings were saved, and no star ratings were synced to cloud.
Also the title-bar * symbol indicating unsaved content was not updated at the beginning of the sync as I would have expected if a save were actually performed, so I can pretty much guarantee that you don't do save to disk before a sync when cloud storage is online.

mturkia, 2020-09-06:
Can you send us the following logs after a crash, as there is a small change that these might contain some useful information:

%APPDATA%\Subsurface\subsurface_out.log
%APPDATA%\Subsurface\subsurface_err.log

(%APPDATA% should expand to C:\Users\Username\AppData\Roaming, or whatever this directory happens to be in your environment)

sveinmagnus, 2020-09-06:
Hi again,
I'm not sure if the logs might have been wiped since I reopened Subsurface and did a new correction and sync after the crash, but I have attached the logs that were present at least.
[Subsurface.zip](https://github.com/subsurface/subsurface/files/5179357/Subsurface.zip)

mturkia, 2020-09-06:
There is something in the logs that seems odd to me. I just haven't looked into these logs in ages, so not sure if this is common, or a sign of problems. I am sure @dirkhh will be able to tell more. Anyway, grab new logs, if you encounter another crash. Hopefully something useful will be flushed to the disk. (I think these were from the successful run.)

These snippets are from the subsurface_err.log:
`
INFO: git storage: branch sveins@tihlde.org is missing in local repo
exclusive write access obtained...closing handle!rename failed: 5
INFO: git storage: problems with local cache, moved to C:\Users\Svein-Magnus\AppData\Roaming\Subsurface/cloudstorage/1deea872b7b67001.2
...
INFO: git storage: check remote status
INFO: git storage: branch sveins@tihlde.org is missing in remote, pushing branch
`

The subsurface_out.log contains also something peculiar:
`
rename of C:\Users\Svein-Magnus\AppData\Roaming\Subsurface/cloudstorage/1deea872b7b67001 to "C:\\Users\\Svein-Magnus\\AppData\\Roaming\\Subsurface/cloudstorage/1deea872b7b67001.2" failed
`

BTW you should not put too much trust into the * sign in status bar when analyzing what is being done. The way the cloud storage works is; we save the data to a local git repository and once this is done, we synchronize the local git repository to the one in the cloud. So there is no way we could only save to cloud. The star is cleared only after full save is done, in cloud mode this means after everything is stored also to the cloud repository.

dirkhh, 2020-09-06:
I would suggest deleting the `C:\Users\Svein-Magnus\AppData\Roaming\Subsurface\cloudstorage` directory and trying again. The meta data shows that we have at least one crash in the middle of writing data to the cloud with a partial write to the git backend.

sveinmagnus, 2020-09-09:
Thanks, I have wiped the cloudstorage folder and have not seen any sync errors since. I'll reopen this if I see another sync-crash.

closed 2020-09-09 by sveinmagnus

dirkhh, 2020-09-09:
Excellent! Thanks for reporting back and closing the issue - I really appreciate that!