subsurface/subsurface #2942: Lost half of my dive history from cloud

Jarf10, 2020-09-09:
I don't really know how to describe the issue because I have no clue what happend.
First I downloaded new dives from my dive computer on my linux laptop. I synced this with the cloud. The new dives were available on my phone so that was fully up to date. Today I decided that I would also like these dives available on my windows PC. After installing Subsurface and connecting it to the cloud no dives were synced. Then I looked on my laptop and synced, 35 dives were lost then. On my phone I had all dives available but after a sync (I hoped it would upload all dives to the cloud) also on my phone around 35 dives dissapeared.
I have no idea who to contact with this issue so I hope from the deepest of my heart that someone can help me and hopefully restore some dives.

dirkhh, 2020-09-09:
All dives that were ever successfully synced to the cloud should still be there.
Please send me (dirk at hohndel dot org) an email FROM the email address you used for the cloud account and in that email five me explicit permission to access your cloud data. 
I should be able to restore your dives

Jarf10, 2020-09-09:
@dirkhh truly saved the day for me. Sorry for the inconvenience of opening a issue that possibly has nothing to do with the awesome Subsurface application.

closed 2020-09-09 by Jarf10