subsurface/subsurface #2950: Deco Model on graph does not match deco model reported by computer

ryanm101, 2020-09-13:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Once I'm imported a dive from my Perdix AI and I am looking at the graph the Deco model shown above is 30/70 however if i look at the Extra info tab i can see that the deco model i dived was actually 30/75.

The deco model above the graph is the once set in preferences. 

This would be expected behaviour for computers that do not return a GF setting, however when the computer does then that one should take precedence for that dive record.

### Operating system:
Linux (Fedora 32)

### Subsurface version:
Built from source: 4.9.6-72-gbeedea7ccfc

### Steps to reproduce:
Import a dive with a difference GF from the one in preferences.

### Current behavior:
Shows GF set in preferences above graph.

It also seems that the preferences setting displays no matter which record you are on. I find this misleading as it looks like the GF above the graph is the one you dived which may or may not always be true.

### Expected behavior:
If dive computer returns GF then use that above chart else use preferences setting.

Is above the graph the right place to have this info should there perhaps be a field in information that can be edited for VPM-B / GF / <Computer Manufacturer> (e.g. Suunto) (for DCs that use a propitiatory model)

dirkhh, 2020-09-13:
The GF shown is the one used for the Subsurface internal calculations of a deco ceiling.

Works as designed.

closed 2020-09-13 by dirkhh

atdotde, 2020-09-13:
It is on purpose that there is one pair of gradient factors (that you can set in the preferences) to display all dives and not one per dive (which could come form a dive computer if it happens to use a variant of Bühlmann+gradient factors).