subsurface/subsurface #2952: Feature Request: Google Photos Integration

brandonpollack23, 2020-09-13:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:
It would be nice to be able to authorize via Oauth and have subsurface use the google photos api to populate your dives.
Sadly, there is no FUSE available for google photos, so they're all on the cloud only.  We can, however do something like the following:

1) Auth and get an OAuth2 token
2A) On first sync (or lazily) execute a [search](https://developers.google.com/photos/library/reference/rest/v1/mediaItems/search) operation with a [filter](https://developers.google.com/photos/library/reference/rest/v1/mediaItems/search#Filters) on the date and time of the dive.
OR
2B) Have the user supply an album that Subsurface queries via the search function and matches photo times to dives.
3) Create thumbnails for matching photos and store the URL same as currently.

I've been looking into what it would take to contribute this myself, but I figured I might put in the request in case any current maintainers are also interested/have suggestions.

### Operating system:
Any mobile or desktop.

atdotde, 2020-09-14:
Hmm. What would be the scenario in which you intend to use this (as opposed to local images)? At some point, we had some facebook integration (with OAuth) where you could create posts out of your dives. That broke far too often and was not really used by more than a handful of users. So it was eventually removed (replaced by the possibility to export the profile image out of which you can then manually create a FB post if you want to).

Would the idea be to keep local copies of the images? Otherwise, retrieving them from the web each time you select a dive with images would probably slow down responsiveness by a lot.

Honestly, the only possible benefit I could imagine would be for the mobile version: Currently, on mobile, there are no media since we decided including those in the cloud storage of the actual dive log would be far too heavy. So, if google images could be the cloud storage solution for media that could be interesting. But still, there is a chance that it's just not practical.

But by all means: Feel free to start working on this. The facebook code is of course still in the git repo if you want some inspiration. If you have questions, I am sure people will help you.

dirkhh, 2020-12-12:
no activity

closed 2020-12-12 by dirkhh