subsurface/subsurface #2953: Download error Suunto Zoop

exxus, 2020-09-14:
### Describe the issue:
- [x] Bug

Fail to download dives from Suunto Zoop

### Issue long description:
I tried downloading dives form my Suunto Zoop dive computer.
This worked previously with no problems.
Now it fails due to ERROR: Unexpected answer start byte(s).

### Operating system:
archlinux Linux 5.8.8-arch1-1 #1 SMP PREEMPT Wed, 09 Sep 2020 18:59:45 +0000 x86_64 GNU/Linux

Suunto Zoop with USB serial adapter

### Subsurface version:
Subsurface v4.9.6,
built with libdivecomputer v0.7.0-devel-Subsurface-NG ()
built with Qt Version 5.15.0, runtime from Qt Version 5.15.1
built with libgit2 1.0.1

from archlinux repository

### Steps to reproduce:
Try downloading from DC.

### Current behavior:
Download aborts with error message

### Expected behavior:
Download finishes successfully

### Additional information:
log file attached
[subsurface.log](https://github.com/subsurface/subsurface/files/5216627/subsurface.log)

dirkhh, 2020-09-14:
Nothing has changed in the code or with the Zoop firmware in many years.
The cable connection is known to be very flakey, and the Zoop is known to have more serial issues if the battery level is low.
My suggestions would be to check the battery, check the seating (and dryness) of the cable, and try again.

exxus, 2020-09-15:
Thank you for the information.
Interestingly when selecting "save libdivecomputer dump file" and "force downloading all dives", I get a very different result.
See attached log.
All other combinations lead to the above error.
[subsurface_dump.log](https://github.com/subsurface/subsurface/files/5227346/subsurface_dump.log)

dirkhh, 2020-09-20:
It is, but that's not really a useful approach.
It should work just fine - if your connection is good enough for a data dump, then a regular download should finish as well.

exxus, 2020-09-27:
Thank you for the answer.
I tried to download multiple times without success [multiple_attempts.txt](https://github.com/subsurface/subsurface/files/5287988/multiple_attempts.txt), [subsurface_err.log](https://github.com/subsurface/subsurface/files/5287989/subsurface_err.log) .

Only when downloading the dump file, it continues downloading. It sometimes finishes but sometimes it aborts, but after already downloading some data 
[subsurface.log](https://github.com/subsurface/subsurface/files/5287992/subsurface.log).
Without saving the dump file I have no success even after trying at least 10 to 20 times.

dirkhh, 2020-09-27:
I don't have a Zoop available, but I do have a different old Suunto dive computer that uses the same download backend, and I have no problems downloading from it. The error messages clearly indicate a serial communication issue.
I don't know what else to try at this point.

exxus, 2020-09-28:
Ok, I borrowed a different Suunto Zoop from a dive buddy, and using the same usb serial connector, downloading finishes without errors. So this is clearly a problem with my dive computer and not with subsurface. Strangely downloading the dump file worked.
I will try replacing the battery.

closed 2020-09-28 by exxus