subsurface/subsurface #2954: Translation of application menu and native dialogs issue

johan-git, 2020-09-15:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Translation of Subsurface application menu and native dialog buttons doesn't work on macOS build when changing language to other than English. Preferences, Quit, Services, Hide and "about Subsurface" remains untranslated aswell as native dialog buttons like cancel, apply and save. Minor issue but most translated text in UI refers to a button spelled in set language.

### Operating system:

macOS 10.15.6 (Swedish system language)

### Subsurface version:

Subsurface 4.9.6 (also tested with latest dev/test build) using provided dmg.
Tried Swedish, German and French language setting in Subsurface with the same result

### Steps to reproduce:

-Change language setting in Subsurface under preferences/language to a non english language
-Save settings and restart Subsurface
-Application menu "Subsurface" and Quit/Save/Cancel native dialog boxes in UI remains untranslated (ex. Cancel and Save in Dive planner and preferences

### Current behavior:

Translation of application menu and native dialog buttons doesn't occur
[image]

### Expected behavior:

Translation of application menu and dialog buttons

### Additional information:

I decided to do a quick test on Linux (Ubuntu 20.04) to see if the problem/state with the dialog buttons occurs on other platform and compiled Subsurface from source. The translation of UI dialog buttons like save, apply, cancel and quit remains untranslated for some languages ex. swedish, dutch but works for other languages like german, finnish which make me think that the problem might be both QT and platform related and that this might be two separate issues.

### Mentions:

@dirkhh

dirkhh, 2020-09-15:
These texts come from Qt itself, not from our sources. And they should be translated (and they were translated in the past for most of our languages, certainly for German). I need to dig into this some more to see if somehow we are inadvertently not picking up the Qt translation texts (we had that problem quite a while ago). Thanks for the report. Something else for me to struggle with :-)

dirkhh, 2020-09-15:
I found the problem. And I can't believe we had this bug for such a long time.
Please try the latest development build - either from source or using the DMG in the downloads/test folder.
I tried this on Linux and Mac with a few languages and it seems to work.

closed 2020-09-15 by dirkhh

dirkhh, 2020-09-15:
Ugh - should have played with this some more before pushing to master. I know not to do that. I never do that. Arg.

Yeah, there will be a follow up to fix things for languages that are still using the old pattern.
Swedish and Portuguese for example only have a qt_XX translation and no qtbase_XX translations.
And of course this will never work in Dutch since there is no Dutch Qt translation :-(

johan-git, 2020-09-15:
That was quick, thanks for looking in to it, and great work! Did some testing on macOS and Linux and it looks much better for languages that have qtbase_XX.  When checking some of the nordic countries I can see that Finnish now works as expected on both macOS and Linux, but Danish (which have the qtbase_da and what I can see the same files as fi) dont get the menus and buttons updated on macOS. But what I can see the Danish translations is behind. So this might be something to put in the backlog

commit 1ed8cb37 (2020-09-15): translations: use the right Qt translations and try harder to find them

We were still using the 'qt' translations instead of 'qtbase' as we should have
been using since forever - it's a little unclear from reading the documentation
when in the Qt5 life cycle this happened, but definitely several years ago.
These are the strings used in situations where Qt already provides us with text
(e.g., the entries in the 'Subsurface' menu on Mac, or the button labels in
many dialogs).

Additionally we didn't try hard enough to find those translations in cases
where they are bundled with the app; so basi [... cut]