subsurface/subsurface #2956: Sharewater Perdix AI not connecting

wjrlock, 2020-09-15:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The Perdix won't connect to Subsurface. I get an error "failed to connect" i did try on 2 computers and various installations. I use the original Sharewater BT USB and BT from a HP laptop. In both cases i can connect with the original software and download the dives to the computer but when i use Subsurface, i get an error. 

### Operating system:

Windwos 10 Enterprice and Windows 10 pro 

### Subsurface version:

4.9.6

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-09-20:
There is a fairly detailed explanation of Windows BLE woes as part of the user manual.
The short suggestion is to manually unpair the Perdix AI from Windows, then re-pair it.
Also, ideally make sure that you do not switch between different apps and different device. The Shearwater BLE stack is a bit fragile and gets confused if it interacts with different computers / phones.
This is known to work, our Windows code was actually written by a developer with a Perdix AI

tesposito4, 2020-12-02:
I'd like to add my findings on this to hopefully help other Shearwater owners.  A few (maybe 5?) releases ago, removing and re-adding my Perdix via the Win 10 Bluetooth settings (the Metro UI version? first image below) became very temperamental and maybe 2 releases ago, I couldn't get Subsurface and my Perdix to work together at all via that interface.  However, using the old "Add Device" in the control panel (second screenclipping below) seems to, knock on wood, work every time if I remove and then re-connect my Perdix there. 

Of course, it's Windows so your mileage may vary. 
[image]

[image]

dirkhh, 2020-12-02:
Thank you so much for posting this -- the Windows / BLE experience is quite frustrating and getting suggestions and recommendations like this from other users is extremely helpful!

ryanm101, 2021-05-18:
To be fair it's not much better on Fedora 3x, I have to unpair and repair with my XPS 13 then use Subsurface in forced BLE (not sure if that bit helps but doesn't hurt) every time.

mikeller, 2026-10-17:
Hi @wjrlock,

Thanks for the report. This matches a known Windows BLE pairing issue with the Perdix AI, the Windows BLE support was written by (and tested with) a developer who personally uses this exact device, so it's known-working in principle. Two other users (tjesposito, ryanm101) confirmed the standard unpair/re-pair workaround resolved the same symptom for them; details are in the user manual's Windows BLE troubleshooting section.

Closing this citing that existing guidance and the two independent confirmations it works. Please reopen if unpairing and re-pairing doesn't get your Perdix AI connecting.

closed 2026-10-17 by mikeller