subsurface/subsurface #2960: Minor French translation mistake

fmatray, 2020-09-18:
#1954 # Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
"Current" in info tab is translate in french as "Actuel". It should be "Courant".
"Actuel" in french means "a present/existing thing". 

### Operating system:
Mac OS

### Subsurface version:
4.9.6

dirkhh, 2020-09-18:
French has the by far largest translator group - would you like to contribute to that as well?
It's really easy, Transifex allows login with your GitHub account:
https://www.transifex.com/subsurface/subsurface/translate/#fr_FR/subsurface4
you'll have to send a request to translate Subsurface (which I'll of course approve) and then you can update the translations that are 'off'.
Thanks so much!

fmatray, 2020-09-18:
Hello,
I would be pleased. I just register and sent a request.

Thanks.

dirkhh, 2020-09-18:
(sorry it took so long to approve that - Transifex sends notifications only every NN hours, and makes it surprisingly hard to go in and manually approve pending requests, but I just did that)

fmatray, 2020-09-18:
Correction done and I'll try to work on missing translations during the week-end.

closed 2020-09-20 by dirkhh