subsurface/subsurface #2961: Dive removed from dive list when updating Current or Visibility

fmatray, 2020-09-18:
### Describe the issue:
- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I change Visibility or Current in info tab, the dive disappear from the dive list.
I don't know if it's only a display bug or if divelog is saved without the missing dive.

### Operating system:
MAC OS

### Subsurface version:
4.9.6

### Steps to reproduce:
1) Select a Dive
2) Click on "Information"
3) Change Current or Visibility

### Current behavior:
Selected dive disappears.

### Expected behavior:
It would stay :-)

### Additional information:

### Mentions:

fmatray, 2020-09-18:
I also tried with Atm pressure/Altitude with the same result.

dirkhh, 2020-09-18:
Ugh, that's bad.

I just tried this with the latest development build (as that's what I have easily available on my system); modifying a dive there doesn't make it disappear.
Encouraged by this I tried the official 4.9.6 build for macOS and there, too, I can make sure edits without making the dive go missing.
Now I'm confused.
Can you create a screen video (QuickTime Player makes that easy) that shows the effect? Feel free to share it just with me (dirk at hohndel dot org) if you don't want to post it here. I'm trying to understand what's going on so we can fix it. But without being able to reproduce it, that's pretty hard...

fmatray, 2020-09-18:
I sent you a video screenshot on your email.

I guess it's just a display problem, with some kind of filter not refreshed correctly.
If I go on the tab "Dive sites" and select all dives, it appears again. But when I try to unselect, all dives disappear.

Is there a way to refresh the dive list, please ?

dirkhh, 2020-09-18:
So the video seems to imply that you have an active filter going. Can you hit Command-F and disable any filters that may be active? That's the only way I can explain this.

fmatray, 2020-09-18:
I tried several times and I get the same result.

I tried the reset button and manually checking inputs.

dirkhh, 2020-09-18:
@bstoeger I don't get this one. How can an edit on the info tab suddenly switch us into 'filtered' mode???

bstoeger, 2020-09-19:
>  I don't get this one. How can an edit on the info tab suddenly switch us into 'filtered' mode???

Are you sure about that analysis? Can I see the video?

Note that closing the filter is not enough - you have to reset it.

bstoeger, 2020-09-20:
I can partially reproduce this - when making a dive go away, the a dive is made current, but not selected in the list. Edits are then not registered. I will look into this.

> I tried the reset button and manually checking inputs.

To reset the filter you don't have to click the (x), but the "reload" arrow. I believe that you have an active filter. I reckon we should be more aggressive at displaying an "active filter". I don't understand though how this can be triggered by editing the current - we don't filter for that, do we?

bstoeger, 2020-09-20:
I think I can also reproduce the other part of the problem: when opening a file the current filter is not applied, at least for XML. Only when editing a dive, the filter is checked. Thus, when editing the current, the dive may disappear owing to a *different* filter condition.

dirkhh, 2020-09-21:
Would you be able to try a new test build that should fix the problem?
https://subsurface-divelog.org/downloads/test/Subsurface-4.9.6-121-gf42dee8ac2e4.10.15+11.0.dmg
(you didn't say which version of macOS - if you are on 10.13 or 10.14 there's a different DMG for those: https://subsurface-divelog.org/downloads/test/Subsurface-4.9.6-121-gf42dee8ac2e4.10.13+14.dmg)

bstoeger, 2020-09-29:
Please check new release. The UI should now behave nicer.

fmatray, 2020-09-30:
Sorry, I had a lot of work theses days.

It's ok for me on MacOS 10.15.6.

dirkhh, 2020-09-30:
Awesome. Thanks for the update

closed 2020-09-30 by dirkhh