subsurface/subsurface #2966: No translation of options for water type/density in informatio tab (used to work)

johan-git, 2020-09-21:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When manually adding a dive in a non english language, the translation of the options in water type/density remains untranslated (e.g. Brackish, Salt, Fresh, Use DC) This used to work and I tracked it back to working in 4.9.4 but has been broken in 4.9.5 and 4.9.6 (tested diffrent non-english languages with the same result)

### Operating system:

macOS Catalina 10.15.6

### Subsurface version:

Latest development build, but verfied the issue in 4.9.5 and 4.9.6.
The translation works in 4.9.4.

### Steps to reproduce:

- Change to a non english language
- Restart Subsurface
- Manually add a dive
- In the information tab,  open the drop down menu of water type/density and verify that the text remains untranslated

### Current behavior:

No translation 
### Expected behavior:

Translation of options (text) in drop down menu
### Additional information:

### Mentions:

@dirkhh

johan-git, 2020-09-21:
Another related issue is that the drop down menu is replaced with the text "unknown" for what I can figure out dives that is imported from DC (not manually added). The option to edit the watertype/denisty for a dive isn't possible for these dives in a later stage, at current build. This issue also exists in 4.9.4.

dirkhh, 2020-09-21:
I can see how the bug (untranslated strings) was introduced. I'll fix that for 4.9.7.
Your second concern is actually a preference setting. In the Preferences, under Dive log, the final option enables you to always edit salinity

johan-git, 2020-09-21:
Oh I can see how I missed the setting, doing the testing on the default configuration of Subsurface, my bad, was a little quick when verifying the first problem. Big thanks for your commitment and for fixing the bug

dirkhh, 2020-09-23:
fixed in master and in 4.9.7

commit b6a18f4c (2020-09-21): translations: initialize water type strings at run time

We cannot initialize strings with translated texts before the translations are loaded.
This was introduced in commit 95e6792c4f ("Grantlee: Add salinity and water
type to grantlee variables") and wasn't caught in the code review. Oops.

Fixes: #2966
closed 2020-09-23 by dirkhh