subsurface/subsurface #2971: Subsurface-mobile: adding a dive manually with date in the past causes wrong dive to be shown after saving

dirkhh, 2020-09-21:
### Describe the issue:

- [x] Bug

### Issue long description:

Manually create a new dive, change the dive date to be in the past, before existing dives in the log
save the dive
notice that the dive isn't shown in dive details

### Operating system:

Tested mobile-on-desktop on macOS, tested on Android

### Subsurface version:

current master 4.9.6.125

### Mentions:

@bstoeger - I know you never use the mobile version... and I'm not sure it has anything to do with the model. But I thought I'd add you just in case... :-)

bstoeger, 2020-09-22:
Happens not only for newly added dives.

dirkhh, 2020-09-22:
I played with this for quite a bit and there's a fascinating challenge... we can select a current dive - but somehow, at some point, we lost the ability to tell the UI to SHOW that dive. I need to figure out (again) how to do that. I am pretty sure that at some point by setting a new current dive that one was "automagically" shown. But that doesn't appear to work anymore.

There is one additional issue with the manual adding of dives where sometimes we appear to add two dives by mistake, one at the current time/date with the default depth and duration, and a second one with the values the user entered. I've run into that twice in my test, but I cannot reproduce it.

bstoeger, 2020-09-22:
I'll look into it later today.

bstoeger, 2020-09-22:
Could you please try commenting out
```
        if (oldCurrent == current_dive)
                return;
```
in
`DiveTripModelBase::currentChanged()`
?

bstoeger, 2020-09-22:
I'm not sure I'll have time to look closer into this today.
Apparently the swipe model doesn't update the current position correctly if dives are moved(?), but we don't send a current-changed signal, because current doesn't changed. It's a bit unfortunate that mobile uses `current_dive` and desktop uses the selected dives.

dirkhh, 2020-09-22:
> Could you please try commenting out
> 
> ```
>         if (oldCurrent == current_dive)
>                 return;
> ```
> 
> in
> `DiveTripModelBase::currentChanged()`
> ?

It's kinda interesting to contemplate what the code in the `SUBSURFACE_MOBILE` part will do when the two are the same, right?

bstoeger, 2020-09-23:
You would of course do something like
```
diff --git a/qt-models/divetripmodel.cpp b/qt-models/divetripmodel.cpp
index 8671a62b6..8a6885ab3 100644
--- a/qt-models/divetripmodel.cpp
+++ b/qt-models/divetripmodel.cpp
@@ -538,9 +538,6 @@ static ShownChange updateShownAll()
 
 void DiveTripModelBase::currentChanged()
 {
-	if (oldCurrent == current_dive)
-		return;
-
 	// On Desktop we use a signal to forward current-dive changed, on mobile we use ROLE_CURRENT.
 	// TODO: Unify - use the role for both.
 #if defined(SUBSURFACE_MOBILE)
@@ -549,11 +546,14 @@ void DiveTripModelBase::currentChanged()
 		QModelIndex oldIdx = diveToIdx(oldCurrent);
 		dataChanged(oldIdx, oldIdx, roles);
 	}
-	if (current_dive) {
+	if (current_dive && oldCurrent != current_dive) {
 		QModelIndex newIdx = diveToIdx(current_dive);
 		dataChanged(newIdx, newIdx, roles);
 	}
 #else
+	if (oldCurrent == current_dive)
+		return;
+
 	if (current_dive) {
 		QModelIndex newIdx = diveToIdx(current_dive);
 		emit currentDiveChanged(newIdx);
```
Nevertheless, it is conceptually wrong to send a changed signal for something that hasn't changed and the model shouldn't lose track of the current item if it is moved.

dirkhh, 2020-09-23:
Yes, that should work. Will you send a PR? After all, you fixed the bug :-)

bstoeger, 2020-09-23:
Feel free to commit, if this works for you. I wouldn't call it "fixing" the bug, but rather "hiding".

commit 038f98d0 (2020-09-23): mobile/dive-list: correctly update view when changing dive date

If the dive timestamp changes, the dive could move in the dive list. But the
current dive actually doesn't change (it's still the same dive, right?). Yet
we need to update the dive list as well as the shown dive (especially if this
is after adding a dive, which is first inserted with the current time and then
updated with whatever the user enters).

Fixes: #2971

Suggested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
commit 749182cc (2020-09-27): mobile/dive-list: correctly update view when changing dive date

If the dive timestamp changes, the dive could move in the dive list. But the
current dive actually doesn't change (it's still the same dive, right?). Yet
we need to update the dive list as well as the shown dive (especially if this
is after adding a dive, which is first inserted with the current time and then
updated with whatever the user enters).

Fixes: #2971

Suggested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
PR #2973 (2020-09-27): mobile/dive-list: correctly update view when changing dive date
If the dive timestamp changes, the dive could move in the dive list. But the
current dive actually doesn't change (it's still the same dive, right?). Yet
we need to update the dive list as well as the shown dive (especially if this
is after adding a dive, which is first inserted with the current time and then
updated with whatever the user enters).

Fixes: #2971

Suggested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>

### Describe the pull request:

- [x] Bug fix

### Pull request long description:

This was first noticed whe [... cut]
commit a469dfa3 (2020-09-27): mobile/dive-list: correctly update view when changing dive date

If the dive timestamp changes, the dive could move in the dive list. But the
current dive actually doesn't change (it's still the same dive, right?). Yet
we need to update the dive list as well as the shown dive (especially if this
is after adding a dive, which is first inserted with the current time and then
updated with whatever the user enters).

Fixes: #2971

Suggested-by: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
closed 2020-09-27 by dirkhh