subsurface/subsurface #2974: Cannot find shearwater peregrine on Android

vondwater, 2020-09-24:
### Describe the issue:

- [x ] Bug

### Issue long description:
My Shearwater Peregrine is not found on Android. I have already tried to turn on 'Show all bluetooth devices' toggle in the Advanced settings but still don't see my DC.

### Operating system:
Android

### Subsurface version:
Subsurface version 3.0.15 (4.9.6.111) official release

### Steps to reproduce:
1) Turn on bluetooth connection on DC
2) Try to find DC in subsurface

### Current behavior:
My DC is not found in the app

### Expected behavior:
DC is found in the app

dirkhh, 2020-09-24:
Please use a free app like nRF Connect to connect to and bond with the dive computer.
Then start Subsurface-mobile again. I am 99.9% sure that things will simply start working for you.

vondwater, 2020-09-25:
Thanks for your answer. I have tried to do that but the perigrine is not bondend. How can I bond?
[image]

dirkhh, 2020-09-25:
click on the three dots next to 'DISCONNECT' near the top right and the option will be in that menu.

BLE is unfortunately a very poorly spec'ed protocol. Let me rephrase that - there are great specs, but they leave out a lot of things that would have been important to define. As a result, every dive computer implementing BLE is slightly different and in the end that makes our lives harder. Subsurface-mobile on Android currently isn't looking for BLE devices itself, it relies on the OS telling us about them. That is, honestly, a design flaw that we need to fix. And it's the reason for this odd requirement to bond the dive computer first (that way it will be listed by the OS when we look for it).

Not ideal, but this is a one time process and after that things should "just work" (famous last words)

vondwater, 2020-09-29:
I have tried to do what you wrote but I cannot bond with my peregrine with the app. Is there any other possibility?

dirkhh, 2020-09-29:
Frustratingly this worked a couple of weeks ago, and now it fails for me as well.
Back to the drawing board - I hope I'll have a solution for this soon.

commit 385d31e3 (2020-09-29): Android BLE discovery: use discovery agent

Android can't scan for classic BT devices, so when BT support was first
added, we simply didn't use the discovery agent at all and relied on the
list of paired BT devices provided by Android.

This still worked fine for a lot of BLE devices that allowed 'bonding'
with the Android device - similar to pairing. But some BLE devices (like
the Shearwater Peregrine) don't support bonding and so our Android code
didn't see them at all.

With this commit we start a BLE only scan on Android to add to the list
of already paired devices.

Fixes: #2974
PR #2986 (2020-09-30): Android BLE update

### Describe the pull request:

- [x] Bug fix
- [x] Functional change

### Pull request long description:

a couple of cleanups
better debugging information
use the discovery agent to find BLE devices that can't be bonded/paried

### Related issues:

Fixes #2974 

### Mentions:

@torvalds
commit 4608beae (2020-09-30): Android BLE discovery: use discovery agent

Android can't scan for classic BT devices, so when BT support was first
added, we simply didn't use the discovery agent at all and relied on the
list of paired BT devices provided by Android.

This still worked fine for a lot of BLE devices that allowed 'bonding'
with the Android device - similar to pairing. But some BLE devices (like
the Shearwater Peregrine) don't support bonding and so our Android code
didn't see them at all.

With this commit we start a BLE only scan on Android to add to the list
of already paired devices.

Fixes: #2974
closed 2020-09-30 by dirkhh