subsurface/subsurface #2980: Garmin MK1 Import - Location as Entry or Exit point?

brysconsulting, 2020-09-29:
### Describe the issue:

- [x] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When importing dive data from Garmin MK1, should the Subsurface geo coordinates LOCATION value come from water Entry or Exit field? We see coordinates to be importing as the exit location, it would be better to record the Entry location instead.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

torvalds, 2020-09-29:
Subsurface imports both, and will use the entry point as the main GPS location for the dive.

The exit point will only be visible as "GPS2" in the "Extra Info" tab of the dive (with the entry point as "GPS1")

HOWEVER.

The most common problem with the Garmin Descent is that you don't have any GPS information for the entry point at all, because you didn't go into dive mode and wait for the GPS indicator (the ring around the screen) to turn green. It can take up to a minute.

So then the Garmin won't have an entry point at all, because it never got a GPS lock before you jumped into the water.

And if that happens, subsurface will obviously use the exit point GPS value for the dive location.

In other words, the solution is to always go into dive mode before going into the water, and waiting to make sure you have GPS lock on the Descent.

brysconsulting, 2020-09-29:
Looks like my data in Subsurface is not matching your description. Please see attached screenshots. 

 

Yellow Marking – looks like the application is taking the GPS2 value and not GPS1.

 

Red Marking – if there is no Exit location value, the Subsurface Exit location is recorded with an erroneous value -0.8272…, -08727… 

 

 

 

Regards,

Martin Brys

brysconsulting, 2020-09-29:
[image]

Looks like my data in Subsurface is not matching your description. Please see attached screenshots. 

Yellow Marking – looks like the application is taking the GPS2 value and not GPS1.

Red Marking – if there is no Exit location value, the Subsurface Exit location is recorded with an erroneous value -0.8272…, -08727…

torvalds, 2020-09-29:
Strange. This is our code to parse that thing:

                if (!strcmp(data->key, "GPS1")) {
                        parse_location(data->value, &res);
                        /* If we found a valid GPS1 field exit early since
                         * it has priority over GPS2 */
                        if (has_location(&res))
                                break;
                } else if (!strcmp(data->key, "GPS2")) {
                        /* For GPS2 fields continue searching, as we might
                         * still find a GPS1 field */
                        parse_location(data->value, &res);
                }

ie if we find a GPS1 case, we should take that and stop looking for anything else.

I will have to investigate.

torvalds, 2020-09-29:
Oh, I see what may be going on.

The divecomputer download will try to create a dive site for _any_ string that starts with "GPS". And that code will just associate the last found one with the dive.

That divecomputer download code was actually the original code, and it turns out the new code that prefers GPS1 is essentially never invoked, because it's only activated from "createDiveSite". Which I think you haven't even triggered, because your screenshot shows those original divesites generated by the download.

(And I suspect you can also avoid it by just editing the name on an existing divesite, so even if you do manual editing, that will just keep the old GPS2 values).

What a crock. And the obvious fix to just remove the old code in favor of the new is likely to break something, because we've never tested that situation.

brysconsulting, 2020-09-29:
Sounds like you got to the bottom of this. 

Let me know if you need anything else from me to help with the issue. And if you have the beta release I could do a test import run on my side to confirm the resolution.

torvalds, 2020-09-29:
Ok, I have an untested fix. It looks ObviouslyCorrent(tm).

https://github.com/subsurface/subsurface/pull/2982

dirkhh, 2020-09-29:
@brysconsulting - what OS are you on? I can create a test build for you.

torvalds, 2020-09-29:
That red marking of yours looks odd too, can you select that dive and show the "extra info" for it?

I suspect it's one of those odd "Lap/Record/Session GPS" locations (I don't know what they are - I did all the Garmin FIT file parsing by trial and error), but we *shouldn't* even have picked up on them, because only things that _start_ with "GPS" should trigger our "this is a location" logic.

brysconsulting, 2020-09-29:
Attached is the Extra Info tab with the erroneous GPS data.

[image]

brysconsulting, 2020-09-29:
@dirkhh I am using Windows OS

dirkhh, 2020-09-29:
Can you try this binary?
https://subsurface-divelog.org/downloads/test/subsurface-4.9.7-11-g58a306ba1a79.exe
create a new dive log (Ctrl-W) and download from your Garmin. Let us know how this works out, please.

brysconsulting, 2020-09-29:
This new binary worked much better. See the attached image capture. We also do not see additional Extra Info GPS values, just GPS1 and GPS2.

[image]

brysconsulting, 2020-09-29:
Looks like there is a still small issue with the import. If the ENTRY LOCATION has NO GPS value in the Garmin file, you are using the GPS value from the PRIOR dive in the import sequence. It is possible that variables are not cleared in the code after each dive import, and the GPS data is repeated is subsequent dive.

torvalds, 2020-09-29:
Ok, the erroneous GPS data case is just odd. It should now pick GPS1 (which will have the right value), but GPS2 in there was garbage. 

*Odd* garbage at that. 

It looks like subsurface now does the right thing, but maybe there's some odd libdivecomputer parsing problem for the Garmin. That may be related to the "prior data" issue too.

I will try to figure it out, but nothing obvious comes to mind.

torvalds, 2020-09-29:
Oooh. I see what is going on.

It is indeed a libdivecomputer problem, and it's one that I introduced when I converted my Garmin Descent parser to use the generic field cache. The bug is that it used to clear all the cached data, but after that conversion it only clears the generic field cache data in between dive downloads, so all the Garmin-specific stuff ends up being cached in between dives.

Including, very much, the GPS data.

It's easy enough to fix the stupid brute-force way, but let me think about possible clean alternatives. 

Regardless, I'll have a libdivecomputer pull request shortly - ugly or not.

torvalds, 2020-09-30:
@dirkhh I've updated the libdivecomputer branch with what I think it the fix. 

I did *not* update subsurface, with it, but I guess I can make a pull request for that too. My track record for pull requests that update libdivecomputer is abysmal, though.

torvalds, 2020-09-30:
> I did _not_ update subsurface

I've done so now, and https://github.com/subsurface/subsurface/pull/2989 should fix this all. Closing the issue, let's see if it stays closed this time ;)

closed 2020-09-30 by torvalds

brysconsulting, 2020-10-01:
@dirkhh let me know if/when you have a Windows binary ready for testing