subsurface/subsurface #2994: subsurface mobile doesn't save/record cylinder type

emmdarakis, 2020-10-03:
### Describe the issue:
- [x ] Bug

### Issue long description:
My aim is to get SAC info from dives imported directly to subsurface mobile. 
I import the dive (from my divecomputer.eu unit if that matters) and then edit it to add cylinder type (from the drop down list) and key in start and end pressures, followed by saving the dive. SAC should then be calculated - it is not. I can see that pressure values are saved, but the cylinder I chose is not (the corresponding field is empty). Hence somehow cylinder info is not saved/recorded. This happens every time with dives imported directly from the computer to subsurface-mobile.
Interestingly dives (with cylinder and pressures info) synced from the desktop app to the mobile one can then be edited (eg change cylinder type) in the mobile app and the changes are saved/recorded successfully. 
Cheers

### Operating system:
Android 7.1.2 / MIUI Global 11.0.2
Xiaomi Redmi 4X

### Subsurface version:
Subsurface-mobile 3.0.15
Issue has been happening for some time now (few months maybe)

dirkhh, 2020-12-12:
I cannot reproduce this.
Can you try to give me step by step instructions how to provoke this error?

nano-o, 2021-02-11:
I ran into a similar issue on iOS, although the behavior was not exactly the same. 

Here's what happened:
I installed subsurface on iOS and imported dives from my i300c using bluetooth. At this point cloud sync was off and nothing had ever been synced. I tried editing a dive, and the UI behaved really strangely: every time I edited a dive, the cylinder was changed to the next in the list, without me touching the field at all. Moreover, fields that I changed by selecting from a drop-down menu (like the buddy field) weren't saved; but if I typed the buddy name instead of selecting it from the menu, then it was saved.
Once I enabled cloud synchronization the behavior disappeared and editing dives seems to be working properly now.

I'll try to reproduce next time I import a new dive.

Version: 3.1.2(4.9.10.139)
iOS 14.3

emmdarakis, 2021-04-24:
Sorry for not replying earlier. Thanks to covid we were not allowed to dive since October or so and I totally forgot about this.
The problem still exists.
This morning I download a dive from DC to subsurface mobile. Then I edited the dive to add cylinder, mix and start/end pressures (see screenshot):
[image]
I then press the save button and I get back to dive profile:
[image]
(Notice that SAC is not shown)
I press edit again and now I get:
[image]
Notice that the cylinder is gone.

Not sure if this is related to the problem nano-o describes above.

All these with Subsurface mobile version 3.2.1(5.0.1) running on a Xiaomi Redmi Note 9 Pro/ MIUI Global 12.0.2/ Android: 10 QKQ1.191215.002

closed 2026-09-09 by mikeller