subsurface/subsurface #3009: Qt GUI Style on Windows broken

janmoelter, 2020-10-08:
### Describe the issue:

- [x] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Ever since version 4.8.4 and, consequently, since the transition from Qt 5.9.2.0 to Qt 5.10.1.0, the GUI controls on Windows appear in the old style from the time of Windows 2000, while before they appeared in a modern style. Discussions on the internet suggest that this is because there are libraries missing that have to be included in the build since Qt 5.10. More specifically, this seems to be connected with the QWindowsVistaStyle class providing native GUI controls on Windows that has become a plugin at some point.

### Operating system:

Windows 10, Version 2004

### Subsurface version:

4.9.7; however, this has been an issue in all versions since 4.8.4

### Current behavior:

The Qt GUI controls appear in the old style from the time of Windows 2000.

### Expected behavior:

The Qt GUI controls should adopt a modern style.

dirkhh, 2020-10-08:
I think this is a sign that none of the currently active developers ever actually **USE** the Windows binaries.
I hadn't even noticed, and no one had pointed it out to me.
@neolit123 -- do you know which modules we are missing? Or should I try and figure out what to add? I bet all this needs is an addition to the installer `.nsi` file...

Thanks @janmoelter for pointing this out!!!

neolit123, 2020-10-08:
i actually haven't noticed this as a problem, probably because my other Windows 7 Desktop looks like Windows 2000.

reading on the web the suggested solution is to add `styles/qwindowsvistastyle.dll` in the install package contents
https://doc.qt.io/qt-5/windows-deployment.html
https://stackoverflow.com/questions/54517367/why-qt-user-interface-looks-different-in-release-mode-than-in-debug-mode

this did not work for me, but that may be because i'm mixing Qt DLL versions (i don't have the Qt 5.1x qwindowsvistastyle.dll)

...or maybe this now has to be done explicitly:
https://stackoverflow.com/a/27702836

> since the transition from Qt 5.9.2.0 to Qt 5.10.1.0,

i had a old build of Subsurface and indeed it looks Vista like.

one question here is - given Subsurface has a lot of Windows users but nobody complained about the window style, should this be changed?

neolit123, 2020-10-08:
> i had a old build of Subsurface and indeed it looks Vista like.

ok, so i'm confused...

here is a screenshot of what i get from the official installer running on Windows 10:
[image]

this looks fine to me - i.e. themed like the rest of Windows 10.

neolit123, 2020-10-08:
i also managed to build Subsurface against Qt 5.15.1 and `styles/qwindowsvistastyle.dll` doesn't seem to change anything in my tests.

dirkhh, 2020-10-09:
I never use Windows so I can't tell if this looks "right" or how it "should" look (I considered uploading a Mac screen shot and saying something smart-assy, but you get my point). I'll wait for @janmoelter to comment - maybe we are missing something :-)
Thanks Lubomir for jumping on this - much appreciated!

janmoelter, 2020-10-09:
Here are two screenshots taken on the same computer, one of version 4.8.3 and the other one of the current version 4.9.7.

[image]

[image]

While version 4.8.3 shows flat modern GUI controls similar to the native OS GUI controls, the controls have been in the old style of Windows 2000 as in the current version since then.

For a more direct comparison of the different styles, maybe also have a look [here](https://doc.qt.io/qt-5.15/gallery.html) in the official Qt Documentation.

I also tried to simply add the styles/qwindowsvistastyle.dll library as suggested on the internet. However, this hasn't worked for me even though I managed to get my hand on the Qt 5.15.0 libraries so that their version would match the version of Qt which Subsurface 4.9.7 was compiled against.

Since simply adding the qwindowsvistastyle.dll does not change anything, this seems to suggest that it is not actually "missed". Hence, I suppose it has to be included somewhere in code first. Unfortunately, I am not familiar with Qt at all so that I cannot make a more precise suggestion.

neolit123, 2020-10-09:
thanks for the screenshots. now that i know what to compare i see that the theme is indeed different.

but adding the Qt 5.15.0 qwindowsvistastyle.dll under /styles changed the theme to the desired result:
[image]

this is something that i didn't notice yesterday, since i wasn't comparing the buttons; only the window.

neolit123, 2020-10-09:
works with "plugins/styles/qwindowsvistastyle.dll" too.
fix: qwindowsvistastyle.dll

janmoelter, 2020-10-09:
Awesome! Thanks, @neolit123.