subsurface/subsurface #3014: No way to copy dive

Voludu2, 2020-10-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Manual-described option to copy dive is not available.

### Operating system:
iOS
iPad

### Subsurface version:
3.0.15

### Steps to reproduce:

From dive list
1) long-press on a dive
2) the 3 described options do not appear on the selected dive
3) instead, an Actions panel slides in from right. “Copy dive” is not an option

### Current behavior:
long-press on dive results in “Actions” panel sliding in from right.  There is no action for copying a dive.
### Expected behavior:
 as described in manual: long-press on dive should result in 3 icons overplayed on item in dive list.  One of the icons should be for copy

### Additional information:

[image]

### Mentions:

dirkhh, 2020-10-13:
for some reason we are showing an outdated version of the mobile manual on the website (which is then linked from within the app). I thought I had fixed that, but somehow this got reverted.
The feature that you are looking for has been removed in Subsurface-mobile v3.
I'll keep this issue open until I find time to track down why we have the wrong user manual showing.

dirkhh, 2020-10-13:
I believe the issues is fixed and the v3 manual is once again shown on the website

closed 2020-10-13 by dirkhh