subsurface/subsurface #3020: macOS: Offer possibility to toggle between light and dark theme of Subsurface

hhmeyer, 2020-10-14:
### Describe the issue:

- [ ] Bug
- [x ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

macOS 10.15.7 on iMac

### Subsurface version:

4.9.7 official release

### Steps to reproduce:

Set macOS to the dark theme, start Subsurface

### Current behavior:

Dark theme in Subsurface does not look good or even consistent with macOS.

### Expected behavior:

More pretty ;-)

### Suggestion:
Offer possibility to toggle between light and dark theme of Subsurface in Preferences. I'm fine with the light them of Subsurface even when using the dark them of macOS. I went back to 4.9.5.

dirkhh, 2020-10-14:
It appears that none of the developers use dark theme.
I certainly don't use it - to me, personally, the user experience is terrible and causes eye strain across the board. I am old enough to remember why we all switched to light background in our windows in the 1980s...

I'm happy to entertain changes to Subsurface (i.e., someone sending patches) that don't affect it in regular mode and that help people who want to use dark theme, but I don't see anyone working on that.
As for 'turning it off' - that's explicitly discouraged by Apple. I don't even know if it's possible using Qt. We do quite literally NOTHING to adjust Subsurface or detect if the user is using dark theme.

dirkhh, 2020-11-09:
I think this is significantly improved in the latest builds

closed 2020-11-09 by dirkhh