subsurface/subsurface #3021: Get dive starting time with seconds ?

gonzalobravoargentina, 2020-10-14:
### Describe the issue:

- [ ] Bug
- [ x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I am trying to get the start time of the dive with the seconds included. When I export a dive to a .csv file the time column has this format:  16:54:00 sec . Always show 00 sec. Can I get the exact seconds on this time?

### Operating system:
MAC
 GEO2

### Subsurface version:
4.9.7

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mturkia, 2020-10-15:
When I export to CSV, I get the seconds included. Are you sure the start time of the dive you are exporting is something other than :00?

gonzalobravoargentina, 2020-10-15:
Yes, I download 3 different dives and all of them showed :00 sec

gonzalobravoargentina, 2020-10-15:
The manual of the OCEANIC GEO2 says that the computer register **main time of day** with sec resolution. 

I opened a .uddf file and I have the same issue.  Where can I generate or find the .ssrf file ?

atdotde, 2020-10-15:
I don't know the Geo2, do you download the dives from that computer via the uddf-file? If that is the case and it does not know about seconds, Subsurface obviously has no chance to guess those.

gonzalobravoargentina, 2020-10-15:
I upload dives using USB cable.  The .ssrf  file looks like this:

<settings>
<divecomputerid model='Oceanic Geo 2.0' deviceid='4ea0909b' serial='038616'/>
</settings>
<divesites>
<site uuid='a297547e' name='Folias' gps='-42.794552 -64.942214'>
</site>
</divesites>
<dives>
<dive number='1' date='2020-09-23' time='**16:54:00**' duration='70:50 min'>

gonzalobravoargentina, 2020-10-15:
<settings>
<divecomputerid model='Oceanic Geo 2.0' deviceid='4ea0909b' serial='038616'/>
</settings>
<divesites>
<site uuid='a297547e' name='Folias' gps='-42.794552 -64.942214'>
</site>
</divesites>
<dives>
<dive number='1' date='2020-09-23' time='**16:54:00**' duration='70:50 min'>

mturkia, 2020-10-16:
The time is with minute resolution, so the problem is not with the CSV export. Now the question is, what time the dive computer has, is the problem with it, or is the problem when loading the data from the DC. When you look at the dive log on the dive computer, what is the start time there?

gonzalobravoargentina, 2020-10-16:
When a look to the log in the computer the start time is 16:54 . So It seems to be a issue with the way that the GEO2 storage that time. If this is the problem I will try to get another computer to see i it works.
Could you pass me the models of Computers that you use and work with sec resolution ?
Thanks

atdotde, 2020-10-16:
I just had a look at the libdivecomputer (the library we use to talk to most of the dive computers) source and it seems that there are not too many computers that record the start of the dive with second resolution:

`th-ws-tmpim01:src helling$ grep 'datetime..second' *.c
atomics_cobalt_parser.c:		datetime->second = 0;
citizen_aqualand_parser.c:		datetime->second = bcd2dec(p[0x0C]);
cochran_commander_parser.c:			datetime->second = data[layout->datetime + 1];
cochran_commander_parser.c:			datetime->second = data[layout->datetime + 0];
cressi_edy_parser.c:		datetime->second = 0;
cressi_goa_parser.c:		datetime->second = 0;
cressi_leonardo_parser.c:		datetime->second = 0;
deepblu.c:	data[5] = bcd(datetime->second);
deepblu_parser.c:	datetime->second = 0;
diverite_nitekq_parser.c:		datetime->second = p[5];
hw_frog.c:		datetime->hour, datetime->minute, datetime->second,
hw_ostc.c:		datetime->hour, datetime->minute, datetime->second,
hw_ostc3.c:		datetime->hour, datetime->minute, datetime->second,
liquivision_lynx_parser.c:		datetime->second = array_uint16_le (p + 4);
mares_darwin_parser.c:		datetime->second = 0;
mares_iconhd_parser.c:			datetime->second = 0;
mares_iconhd_parser.c:			datetime->second = 0;
mares_nemo_parser.c:		datetime->second = 0;
oceanic_atom2_parser.c:		datetime->second = 0;
oceanic_veo250_parser.c:		datetime->second = 0;
oceanic_vtpro_parser.c:		datetime->second = 0;
suunto_d9_parser.c:			datetime->second = p[6];
suunto_d9_parser.c:			datetime->second = p[2];
suunto_eon_parser.c:		datetime->second = 0;
suunto_eonsteel.c:	msec = datetime->second * 1000;
suunto_vyper_parser.c:		datetime->second = 0;
tecdiving_divecomputereu_parser.c:		datetime->second = data[7];
`
But on the other hand, I don't think this is too bad: After all, the beginning is not defined too well, most computers define it as the point where you are first at some significant depth (like 1.5m for example), so it's kind of arbitrary.

gonzalobravoargentina, 2020-10-16:
Great info! I will try to find one of those models.

I am interested in getting the starting time with sec in order to transform the dive time (min:sec) into local time and be able to synchronize it with the TIME in a GPS device in the surface.  

> But on the other hand, I don't think this is too bad: After all, the beginning is not defined too well, most computers define it as the point where you are first at some significant depth (like 1.5m for example), so it's kind of arbitrary.

This is another issue that would give me some error, the best escenario would be to get the local real TIME for each dive computer record (each 2 sec)

mturkia, 2020-10-18:
Garmin Descent Mk1, Suunto Vyper Air, Suunto D4 and Shearwater Perdix AI are DCs that I have and record the second when going into dive mode. However, I do not generally see this resolution as critical when mapping GPS location. It usually takes quite a bit of time at the dive site to get ready, so the recorded coordinates are close enough even with one minute resolution on the DC. One exception is when jumping in from a zodiac, but even then I would think that a minute resolution is generally close enough. (BTW on the last dive, it seems that the time on my dive computers is 32 seconds apart when looking at the ones recording it precisely, so you must be much more disciplined than me if you are going to achieve time on GPS tracker and dive computer to be in sync :D)

Note that the list given from libdiecomputer is showing DC families, so each family might contain multiple divecomputers.

gonzalobravoargentina, 2020-10-18:
I didn't explain in detail. I use time to synchronize an underwater photo with the GPS in surface and the next step is to add depth info from the diving computer and to do this  I need to know the exact time during the dive. 

> Note that the list given from libdiecomputer is showing DC families, so each family might contain multiple divecomputers.

Good to know this. I notice there aren't any aqualung models.  I was close to buy  a aqualung i100   :( 

Thanks for all the info.

closed 2020-12-12 by dirkhh