subsurface/subsurface #3030: keyword field error

svebor71, 2020-10-17:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
There is a probem with colored fields like keyword or dive partners on selecting / dragging text.
Some times I select the keyword input, press CTRL+A or select some partial text, the text wanishes leaving empty color bubbles behind. Minimal mouse movements on selecting could although be the reason - so perhaps it is a drag/drop problem, it is not realy clear.
Some times the text comes back on reselect an input, but playing with it also can crash the hole program.
Best way to solve this problem is to restart and hope...

### Operating system:
OS: Windows 10 Pro 2004
Language: german

### Subsurface version:
Official version 4.9.7 and older - german translation

### Steps to reproduce:
most likely on:
1) select some text in an input field with colored text bubbles like keyword
2) drag the selected text to another colored input...
3) ...until the text wanishes or the colored bubbles gets a shadow (an then the text wanishes) 
4) ...keep playing arround with it to crash the program

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2020-10-21:
I can confirm this on Linux. Very obscure. I would have never got the idea to drag text in that way...!

bstoeger, 2020-10-21:
Cool - I managed to freeze my X-Server when running this in the deugger! :-D

bstoeger, 2020-10-21:
Crash always happens in `QTextLine::cursorToX`. Sometimes the X-server doesn't freeze and I get:
```
#0  0x00007ffff6aed25f in QTextLine::cursorToX(int*, QTextLine::Edge) const () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#1  0x000055555577ff8b in QTextLine::cursorToX (edge=QTextLine::Leading, cursorPos=<optimized out>, this=0x7fffffffa830)
    at /usr/include/x86_64-linux-gnu/qt5/QtGui/qtextlayout.h:243
#2  GroupedLineEdit::paintEvent (this=0x55555602dd50, e=0x7fffffffac80) at /home/bs/src/orig/subsurface/desktop-widgets/groupedlineedit.cpp:173
#3  0x00007ffff75372b6 in QWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff75e4d52 in QFrame::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007ffff5e5264b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff74f4a55 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff74fe0f0 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff5e5293a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff752ff4a in QWidgetPrivate::sendPaintEvent(QRegion const&) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff7530799 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff7531306 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff7531306 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff7531306 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff75304a4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff7531423 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*)
    () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...]
#53 0x00007ffff5eaa565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#54 0x00007ffff5e514db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#55 0x00007ffff5e59246 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#56 0x00005555556adae4 in main (argc=<optimized out>, argv=<optimized out>) at /home/bs/src/orig/subsurface/subsurface-desktop-main.cpp:118
```
Ouch.

bstoeger, 2020-10-23:
Some more analysis:

These are at least(!) two bugs, both of which a can reproduce reliably.

1) The crash is in our code. It goes away if I replace the `TagWidget` class by a simple `GroupedLineEdit` stub.

2) The formatting bug is inherent to how `GroupedLineEdit` uses the the underlying `QPlainTextEdit`: By drag & and dropping, you can actually move the "viewport" of the `QPlainTextEdit`. The only way I found to undo that is by drag & dropping again. I wonder if this is a Qt bug?

@dirkhh - help?

bstoeger, 2020-10-24:
More analysis:
Seems to be a Qt/GroupedLineEdit bug after all.

The crash is avoided by the sanity checks done by:
```
diff --git a/desktop-widgets/groupedlineedit.cpp b/desktop-widgets/groupedlineedit.cpp
index fdce55ca4..5639b2e1b 100644
--- a/desktop-widgets/groupedlineedit.cpp
+++ b/desktop-widgets/groupedlineedit.cpp
@@ -161,7 +161,15 @@ void GroupedLineEdit::keyPressEvent(QKeyEvent *e)
 
 void GroupedLineEdit::paintEvent(QPaintEvent *e)
 {
-	QTextLine line = document()->findBlock(0).layout()->lineAt(0);
+	// Do some sanity checks. Without these, the code may crash if
+	// the user manages to scroll the text box by drag&drop actions.
+	const QTextBlock &block = document()->findBlock(0);
+	if (block.lineCount() <= 0)
+		return QPlainTextEdit::paintEvent(e);
+	const QTextLayout *layout = block.layout();
+	if (layout->lineCount() <= 0)
+		return QPlainTextEdit::paintEvent(e);
+	QTextLine line = layout->lineAt(0);
 	QPainter painter(viewport());
 
 	painter.setRenderHint(QPainter::Antialiasing, true);
```
Ironically, this makes the UI interaction even worse - it semi-hangs the X-server, because for some reason the drag&drop action is not acknowledged. I.e. it stays in a limbo-state until you drop your selection onto a terminal window or similar.

bstoeger, 2020-10-25:
@svebor71: could you please check whether the situation is improved in current master? Note: I explicitly say "improved", not "fixed".

svebor71, 2020-10-25:
@bstoeger: much better! Got not crash any more, but had a formating problem once (color bubbles with bottom shadow and no text displayed). After playing a little in two different dive sets the text reappeared. 
(Tested with subsurface-4.9.7-155-ga58eee9aa39b.exe , 2020-10-24 15:37)

dirkhh, 2020-12-12:
I'll close this as the fix is in the current release.

closed 2020-12-12 by dirkhh