subsurface/subsurface #3057: Segfault on unset DCs

salvadorcunat, 2020-11-01:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'm having a segfault (shortened bt follows) coming from get_device_for_dc(), which is present at current master b2b3544f3:

#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff569d537 in __GI_abort () at abort.c:79
#2  0x00007ffff5a357ec in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff5a40966 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff5a409d1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5a40c65 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5a37faa in std::__throw_logic_error(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x0000555555816aaa in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*> (
    __end=0xffffffffffffffff <error: Cannot access memory at address 0xffffffffffffffff>, __beg=0x0, this=0x7fffffffd740) at /usr/include/c++/10/bits/basic_string.tcc:206
#8  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<char const*> (
    __end=0xffffffffffffffff <error: Cannot access memory at address 0xffffffffffffffff>, __beg=0x0, this=0x7fffffffd740) at /usr/include/c++/10/bits/basic_string.h:247
#9  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*> (
    __end=0xffffffffffffffff <error: Cannot access memory at address 0xffffffffffffffff>, __beg=0x0, this=0x7fffffffd740) at /usr/include/c++/10/bits/basic_string.h:266
#10 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<std::allocator<char> > (__a=..., __s=0x0, this=0x7fffffffd740)
    at /usr/include/c++/10/bits/basic_string.h:527
#11 get_device_for_dc (table=<optimized out>, dc=<optimized out>) at /home/boret/src/subsurface/core/device.cpp:40
#12 0x0000555555816fa3 in get_dc_nickname (dc=dc@entry=0x555555b12658 <displayed_dive+184>) at /home/boret/src/subsurface/core/device.cpp:198
#13 0x000055555578db54 in ProfileWidget2::plotDive (this=0x5555563870a0, d=0x55559437a8d0, force=force@entry=false, doClearPictures=doClearPictures@entry=false, instant=instant@entry=false)
    at /home/boret/src/subsurface/profile-widget/profilewidget2.cpp:813
#14 0x00005555556aa706 in MainWindow::selectionChanged (this=0x555555cb2500) at /home/boret/src/subsurface/desktop-widgets/mainwindow.cpp:451
#15 0x00007ffff5e708c6 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x000055555570d16c in DiveListView::setSelection (this=0x5555562fcc40, rect=..., flags=...) at /home/boret/src/subsurface/desktop-widgets/divelistview.cpp:491
#17 0x00007ffff771fe39 in QAbstractItemView::mousePressEvent(QMouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff750852e in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff75afd4e in QFrame::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff5e3b87b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff74c802e in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5

It seems to be triggered by dives which lack of a properly defined DC, something like:
```
  <divecomputer>
  <depth max='9.0 m' />
  </divecomputer>
</dive>
```
This is an uncommon situation.  Manual dives have a tagged  model="manually added dives" and doesn't make subsurface crash, but imported dives may be not (as is the current case with SmartTrak imported manual dives, other imports I don't know).
I'm preparing a PR to fix smtk2ssrf tool for future imports, but divelogs including previously imported dives should be affected.

### Operating system:

Linux, and most probably Windows and Mac
### Subsurface version:

Current master b2b3544f3
### Steps to reproduce:

### Current behavior:

### Expected behavior:

4.9.7 behaves well, it doesn't segfault and show an "unknown dive computer" or something like that in the profile window.

### Additional information:

A log file like this will make current subsurface crash:
```
<divelog program='subsurface' version='3'>
<settings>
</settings>
<site uuid='27554589' name='Nederland, Vinkeveen, Vinkeveense plas'>
</site>
<dives>
<dive number='1' tags='Fresh water, No stop, Single ascent' divesiteid='27554589' date='1993-11-08' time='11:00:00' duration='30:00 min'>
  <buddy>A buddy, Another one</buddy>
  <notes>Smartrak Activity: Training, Instruction
Smartrak Gear: Fourth element rashguard
Opleidingsduik PADI OWD module 1</notes>
  <suit>Two piece 7mm wet suit</suit>
  <cylinder size='10.0 l' workpressure='300.0 bar' description='Tank 10 liter Steel 300 bar' />
  <weightsystem weight='0.0 kg' />
  <divetemperature air='0.0 C' water='0.0 C'/>
  <divecomputer>
  <depth max='9.0 m' />
  </divecomputer>
</dive>
</dives>
</divelog>
```
### Mentions:

@dirkhh, @bstoeger

bstoeger, 2020-11-01:
Yes, the device-code was quite heavily changed recently - thanks for testing that. Besides setting the model, we should also do
```
        device dev { dc->model ?: "", dc->deviceid, {}, {}, {} };
```
in get_device_for_dc().

It's somewhat unfortunate that we treat NULL and "" as the same thing.

salvadorcunat, 2020-11-02:
Hi Berthold.
Wouldn't something like
```
40         if (dc->model) {                                                                                                                                                             
41                 device dev { dc->model, dc->deviceid, {}, {}, {} };
42                 auto it = std::lower_bound(dcs.begin(), dcs.end(), dev);
43                 return it != dcs.end() && same_device(*it, dev) ? &*it : NULL;
44         } else {
45                 return NULL;
46         }
```
on the get_device_for_dc() func just do the trick?

bstoeger, 2020-11-02:
Hi Salva,

I think this would change the behavior of the function with respect to what we had in 4.9.7. The old code would gracefully handle empty-model devices. However, I don't know if it is even possible to have such devices in the device list.

salvadorcunat, 2020-11-02:
I've been testing both options, and both work well.
dcText.isEmpty() in plotDive() handles both equally and we get an "Unknown dive computer" in the profile.
Your solution, however, is more consistent, I think.

bstoeger, 2020-11-02:
The difference is that if you have a device without model but with id in your device list, the old code would find it and your proposed code doesn't (still better than a crash, of course). As I said, I'm not sure that this situation is even possible. However, better try to not change behavior...

bstoeger, 2020-11-06:
I reckon we can close this?

salvadorcunat, 2020-11-06:
Sure Berthold.
Best Regards.

closed 2020-11-06 by salvadorcunat