subsurface/subsurface #3068: Translation of undo command when editing cylinders in equipment tab is not working

johan-git, 2020-11-05:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When adding/removing/editing cylinders in equipment tab the translation of the undo command (Edit in main menu) doesn't work as expected. The word "Undo" gets translated but not the rest of the string which describes the command. For example:
"Undo Add cylinder" doesn't translate **Add cylinder**" 
"Undo Edit cylinder" doesn't translate **Edit cylinder**" 
Tested on several different languages with the same result. So all non English languages are affected.

### Operating system:

Verified on Linux and MacOS

### Subsurface version:

Using latest development build

### Steps to reproduce:

- Change to a non English language
- Restart Subsurface
- Add a new dive
- Go to Equipment tab
- Add a new cylinder 
- Verify result under **Edit** in the main menu

### Current behavior:

Part of string get translated

### Expected behavior:

Translation of entire string

### Additional information:

### Mentions:

dirkhh, 2020-11-05:
@bstoeger - these are set in the constructor calling `::tr()`, so I'm guessing that we are once again in a situation where the translations aren't loaded when the constructor is executed?

bstoeger, 2020-11-06:
Yes, these need a `Command::Base::` as in the rest of the file.

commit 947a100d (2020-11-08): translations: brute force adjustment of tranlated text

Thanks to commit 299ba20364 ("Update translation source strings") all
the translated terms in the Command classes are now part of the base
class. So remove the references to the subclasses in the translation
sources as well.

Fixes: #3068
PR #3074 (2020-11-09): Fix translations

### Describe the pull request:

- [x] Bug fix

### Pull request long description:

I hate doing this so close to a release, but this makes the translations actually work for me, and they clearly didn't work before, so this seems like a net win :-)

### Changes made:

As @bstoeger suggested, simply use the `Command::Base::tr()` function everywhere. And then, of course, make the corresponding adjustments to the translations themselves.

### Related issues:

Fixes #3068
closed 2020-11-09 by dirkhh