subsurface/subsurface #3076: printing:print preview crashes

Egbertdepauw, 2020-11-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Subsurface crashes when I try to do a print preview. It crashes on Windows and on Linux.
HTML export works on both Windows and Linux, and actual printing works on Windows, but crashes on Linux.

### Operating system:

Windows10 Pro for Workstations build 1909 (18363.1198) NL
Ubuntu 20.10 x86_64 

### Subsurface version:

Subsurface 4.9.8 via downloaded installer
Subsurface 4.9.8 Appimage on Linux

### Steps to reproduce:

- Start subsurface
- Select dive from dive list
- Select menu > File > print
- Select Preview (and also Print on Linux)

### Current behavior:

Subsurface crashes
### Expected behavior:

Display the print preview

### Additional information:

I've tried running "subsurface -v" but nothing is logged to console.

On ubuntu the following is logged in dmesg
`_traps: AppRun[16547] trap divide error ip:52924f sp:7ffe97d629d0 error:0 in subsurface[400000+5c9000]_`

In the Windows Application event log, the following has been written:

> _Naam van toepassing met fout: subsurface.exe, versie: 0.0.0.0, tijdstempel: 0x00000000
> Naam van module met fout: subsurface.exe, versie: 0.0.0.0, tijdstempel: 0x00000000
> Uitzonderingscode: 0xc0000094
> Foutmarge: 0x0000000000028ffb_

### Mentions:

bstoeger, 2020-11-12:
I can reproduce that on Linux. @dirkhh: better wait with the release.

dirkhh, 2020-11-12:
Will do. I had a server die (hardware failure) and am busy trying to get the VMs recovered and re-distributed among the other two systems.

bstoeger, 2020-11-12:
The print preview crashes for me because this line in `Printer::print()`:
```
        dpi = printerPtr->resolution();
```
gives a whopping 21846 DPI. I wonder who has such an amazing printer or display?

The QPainter responsible for drawing the profile then gives up. If I remove the profile-drawing code, everything works as expected. Simply changing the DPI here does not work, because then the preview is a small "stamp" at the top of a huge page.

dirkhh, 2020-11-12:
this is code that needs some TLC.  A lot of hacks have been layered on top of each other because the code is used in three very different ways: live profile on desktop, print of the profile, dead but zoomable profile on mobile.
The thing needs to be killed and rewritten (not the profile itself, the way it's used in these three scenarios).
On my very very long list of things that I wanted to look at after the release.
I guess now we are looking at it before 4.9.9?

bstoeger, 2020-11-12:
That was a symptom, not the problem. The problem was uninitialized and therefore random DPI values - I have no idea how this ever worked.

But yes, the printing code has room for optimization.

Egbertdepauw, 2020-11-12:
I've tested it by setting the resolution :
`printerPtr->setResolution(600);`

And now it seems to work. 

Shouldn't the resolution be initialized with the printoptions.ui resolution QSpinbox ?

bstoeger, 2020-11-12:
Please checkout #3079.

dirkhh, 2020-11-12:
should be fixed in 4.9.9

closed 2020-11-12 by dirkhh

Egbertdepauw, 2020-11-13:
Just tested this on my Windows 10 workstation and the preview works again 👍