subsurface/subsurface #3080: application crash in dive planner with 100% oxygen with higher screen width

info-kubu-at, 2020-11-13:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

in diveplanner (Ctrl-L) - add 100% oxygen to gas and resize screen to more than 1600 pixels width. The application crashes.

### Operating system:

Windows 10 20H2 / German Version

PC

### Subsurface version:

Subsurface >4.9.6 
version 4.9.6 is the last version where this works

>4.9.6

official release

### Steps to reproduce:

1) start application, resize screen to lower with of about 1511 pixels
2) use dive planner (Ctrl-L)
3) add pure oxygen (100%) to gas list (99% works)
4) resize application window to a bit more than 1511 pixel width (lets say more than 1700 pixel)

### Current behavior:

application freezes for about 5 seconds and terminates afterwards

### Expected behavior:

no crash

### Additional information:

### Mentions:

[image]

info-kubu-at, 2020-11-13:
[image]

with about 1575 pixels width, it hangs (you can see the message "Keine Rückmeldung" in the title-bar)

Egbertdepauw, 2020-11-13:
Hmm I have tested this, and subsurface crashes, but only when I remove the "O2 is narcotic" checkbox. 

In my case it doesn't crash resizing, but it crashes when I use 100% oxygen and have the "O2 is narcotic" checkbox unchecked.

### Subversion version
Subsurface 4.9.8

@dirkhh Maybe look at this first before releasing v4.9.9 ?

PR #3081 (2020-11-13): Gracefully handle infinite MND for oxygen
When breathing pure oxygen and considering it not
narcotic, there is not maximal narcotic depth and
the formula divides by zero. So better, handle this
case separately.

Fixes #3080

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:
commit 10bedf02 (2020-11-13): Gracefully handle infinite MND for oxygen

When breathing pure oxygen and considering it not
narcotic, there is not maximal narcotic depth and
the formula divides by zero. So better, handle this
case separately.

Fixes #3080
closed 2020-11-13 by dirkhh