subsurface/subsurface #3082: Hangs when starting up with cloud storage (macOS)

calmh, 2020-11-14:
### Describe the issue:

- [x] Bug

### Issue long description:

When starting Subsurface for the first time in a while it hangs on "Sync with cloud storage". Killing it and immediately starting it again resolves the issue (for this time). I'm not sure how long a time needs to pass between invocations to cause the issue; this typically happens about weekly after I've been diving, and it's always the same. Perhaps some cache? The cloud timeout has no effect and there is no relevant error output that I can see:

[image]

This happens on both my Macs so it's not specific to a single machine.

### Operating system:

macOS Big Sur but this happened under Catalina as well. 

### Subsurface version:

4.9.9.

### Steps to reproduce:

Make Subsurface use cloud storage by default. Don't start Subsurface for a long time. Start Subsurface.

dirkhh, 2020-11-14:
Given that I use Subsurface across approximately 20 different systems (almost all of them VMs, of course) and often have longer breaks between testing on a particular system, I would expect that I would have run into this before.
I wonder if Subsurface would tell us where exactly it hangs if you started it with `-v`.
The usual flow is this:
- check that you can reach the server (that's where the timeout comes in)
- pull from the remote git repo into the local cache (that's where you see the progress bar - which I can't see in your screen shot)
There are a number of ways that this can stall, including issues on the server with network timeouts, etc. The server sits on a symmetrical Gb/s internet connection that usually is reasonably responsive, but stuff happens. On the other hand, if this happens to you every single time, then maybe something else is going on.
Oh, how many dives are in your repo? Could this be some git garbage collection (assuming you have thousands of dives)?

calmh, 2020-11-14:
The repo is small, only 40 dives. The funny thing is that it's *just* on the "first" start after ... however long it takes to cause. I tried it now 8 hours after last (when I reported this) and it started up without fault, so longer than that at least. I haven't run with -v before (generally do the GUI kind of startup) but will do that the next few times and hope to maybe narrow it down.

There is in fact a progress bar in the small dialog above, with only a couple of pixels filled in. It becomes invisible in Big Sur when the window isn't focused, due to some GUI toolkit change/incompatibility with Qt...

Since it always works on second startup it feels like some cache becoming primed -- or possibly a race of some kind due to timing being different when DNS etc isn't already cached. Network connection etc when it actually works seems fine, I'm on 1 Gbps at home and cloud sync is smooth apart from this startup wrinkle...

calmh, 2020-11-29:
This hasn't happened for a while now. Given I can't reproduce any more and nobody else has reported it, closing as a one-off weird thing. (Can always be reopened if someone disagrees.)

closed 2020-11-29 by calmh