subsurface/subsurface #3085: Desktop app hangs when saving to cloud storage

djunkins, 2020-11-15:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After making a change to the dive log, the application hangs when updating the cloud storage. It hangs with the "Transfer to storage (0/7)" modal window showing. Once the application is Force Quit, the update that was being saved appears to have been successfully saved. I've waited as long as 5 minutes for the save to finish before I gave up.

### Operating system:

Mac OS 10.15.6 on a MacBook Air.

### Subsurface version:

Subsurface 4.9.9

I first ran into the problem when I had self-compiled Subsurface, but I downloaded 4.9.9 to ensure I did not have a library issue.

### Steps to reproduce:

1) Open Subsurface
2) Edit the notes on a dive.
3) Hit <Command>-S to save to cloud storage
4) Application hangs

### Current behavior:

### Expected behavior:

### Additional information:

[image]

[subsurface-crashlog.txt](https://github.com/subsurface/subsurface/files/5542915/subsurface-crashlog.txt)

### Mentions:

dirkhh, 2020-11-15:
Unsurprisingly that isn't my experience, but then again, the cloud storage happens to be two floors down from where I sit (so technically I'm "above the cloud").
Seriously, though... I'd love to figure out what triggers this. Can you get a verbose log to see which operation actually hangs? It seems like it's the opening of the remote git repo. So that could indicate some weirdness in your local cache.
Have you tried moving your cache dir out of the way (you'll get the location of that in `-v` output as well).

djunkins, 2020-11-15:
I'll try to grab the verbose log and then clear the local cache. It looks like someone else just reported something similar on the Subsurface mail list.

djunkins, 2020-11-15:
Here is the output from running with the -v option
`orcas:MacOS junkins$ ./Subsurface -v
Subsurface v4.9.9,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (0ebd5d3879854206cd1cf0272e6c357004b33727)
built with Qt Version 5.12.1, runtime from Qt Version 5.12.1
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 2.1."
cloud URL set as "https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]"
loading dive data from ("https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]")
git storage: Synchronising data file
INFO: git storage: update local repo

INFO: git storage: sync with remote https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]

git storage: Sync with cloud storage
INFO: git storage: delete proxy setting

Cloud storage: checking connection to cloud server
Checking cloud connection...
QNetworkReplyHttpImplPrivate::_q_startOperation was called more than once QUrl("https://cloud.subsurface-divelog.org//make-latte?number-of-shots=3")
INFO: git storage: fetch remote

git storage: Successful cloud connection, fetch remote
INFO: git storage: certificate callback for host cloud.subsurface-divelog.org with validity 1

INFO: git storage: check remote status

INFO: git storage: try to update

git storage: Done syncing with cloud storage
git storage: Load dives from local cache
git storage: Successfully opened dive data
Set the current dive site: 123054703

File locations:

cloud URL set as "https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]"
Local git storage: /Users/junkins/Library/Application Support/Subsurface/cloudstorage/1f192ae9e4ba4a5f
Cloud URL: https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]
Image filename table: /Users/junkins/Library/Application Support/Subsurface/hashes
Local picture directory: /Users/junkins/Library/Application Support/Subsurface/picturedata/

QNetworkReplyHttpImplPrivate::_q_startOperation was called more than once QUrl("http://mt.google.com/vt/lyrs=s&hl=en-US&x=17&y=9&z=5&s=Gali")
dive # 150 changed, cache is invalidated
Set the current dive site: 123054703
git storage: Synchronising data file
INFO: git storage: update local repo

INFO: git storage: sync with remote https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]

git storage: Sync with cloud storage
INFO: git storage: delete proxy setting

Cloud storage: checking connection to cloud server
Checking cloud connection...
QNetworkReplyHttpImplPrivate::_q_startOperation was called more than once QUrl("https://cloud.subsurface-divelog.org//make-latte?number-of-shots=3")
INFO: git storage: fetch remote

git storage: Successful cloud connection, fetch remote
INFO: git storage: certificate callback for host cloud.subsurface-divelog.org with validity 1

INFO: git storage: check remote status

INFO: git storage: try to update

git storage: Done syncing with cloud storage
INFO: git storage: do git save

git storage: Preparing to save data
git storage: Start saving data
git storage: Start saving dives
INFO: removed reference to non-existant dive site with uuid 0755aa6f

git storage: Done creating local cache
INFO: git storage, write git tree

INFO: existing filename https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]

INFO: Commit message:

Changes made: 

Edit notes [#150]

Created by Subsurface:4.9.9.0:macOS 10.15:x86_64:en-US

INFO: git storage: sync with remote https://cloud.subsurface-divelog.org//git/junkins@foghead.com[junkins@foghead.com]

git storage: Sync with cloud storage
INFO: git storage: delete proxy setting

Cloud storage: checking connection to cloud server
Checking cloud connection...
INFO: git storage: fetch remote

git storage: Successful cloud connection, fetch remote
INFO: git storage: certificate callback for host cloud.subsurface-divelog.org with validity 1

INFO: git storage: check remote status

INFO: git storage: try to update

INFO: git storage: local is newer than remote, update remote
git storage: Push local changes to cloud storage
INFO: git storage: update remote

INFO: git storage: certificate callback for host cloud.subsurface-divelog.org with validity 1

INFO: git storage: certificate callback for host cloud.subsurface-divelog.org with validity 1

git storage: Transfer to storage (0/7)
^C
orcas:MacOS junkins$ 
`

Removing the local cache did not improve the situation.

dirkhh, 2020-11-15:
Hmm. I did what every decent sysadmin recommends in situations like this. I rebooted the cloud server.
Can you try again? I've tried it forcing an external cloud connection (tethering through my phone) and cannot reproduce the problem.

djunkins, 2020-11-15:
We still live in a <ctrl><alt><del> world...

I am not longer having the issue. Unfortunately I tried one other thing (given the INFO message "INFO: removed reference to non-existant dive site with uuid 0755aa6f" in the log) and purged unused dive sites from the Dive Site tab before trying again, so I'm not sure which fixed the problem.

dirkhh, 2020-11-15:
Extremely unlikely. That's a harmless warning

djunkins, 2020-11-23:
I've not seen any further problems since Dirk reset the cloud storage. Closing the Issue.

closed 2020-11-23 by djunkins

dirkhh, 2020-11-23:
Thanks for the update!