subsurface/subsurface #3087: Current and Visibility stars are cutoff

johnsgill3, 2020-11-18:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The 5 star rating for Current and Visibility are cutoff on Mac

### Operating system:

Mac OS X Big Sur

### Subsurface version:

4.9.9

### Steps to reproduce:

1) Go to Information
2) Look at the Current and Visibility Section
3) The 5th star is cutoff

### Current behavior:

Clicking 5th star is tricky
images.githubusercontent.com/8923546/99589504-ed705400-29a0-11eb-8729-86ea4e488367.png">

### Expected behavior:

Be able to see all/select all stars easily
<img width="212" alt="Screen Shot 2020-11-18 at 1 16 25 PM" src="https://user-

### Additional information:

### Mentions:

dirkhh, 2020-12-12:
So this should change if you make the window wider. Depending on the default font / font size that the OS provides us with, sometimes the default size of our window is too small.
Have you tried this?

johnsgill3, 2020-12-24:
This does in fact change if I make the window space larger. I also adjusted the font and it doesn't have much impact. I guess we can close this resolved.

closed 2020-12-24 by johnsgill3