subsurface/subsurface #3094: Scubapro G2 syncing wrong dive time and can't edited

satoschu, 2020-11-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Last time I synced a dive from Scubapro G2 to my mobile with subsurface mobile 3.1.0 ( 4.9.9.31 ) the dive time was wrong and I was mit possible to edit.
Dive time which was shown 48 min and real time was 57 min.

### Operating system:

Android 9

### Subsurface version:

3.1.0 ( 4.9.9.31 )

### Steps to reproduce:

Syncing a dive

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2020-12-12:
You can edit the dive time (which is when a dive starts), but a dive with profile has calculates its duration based on the profile - therefore you can't edit the duration.

satoschu, 2020-12-12:
But why is there a difference in syncing the dive time?
Syncing in Scubapro Logtrak and the value on the dive computer shows 57min and in Subsurface is 48 min?

dirkhh, 2020-12-12:
Can you post a screen shot of a dive where this is the case? My guess is that you have a surface interval (or an extremely shallow interval) in the middle?

satoschu, 2020-12-13:
Yes, there is a part on the surface in the dive profile.

dirkhh, 2020-12-13:
Well, I guess then we will just have to agree to disagree. The Subsurface developers believe that the duration of a dive is the time that you are actually _diving_. So the dive below, in our mind, is a 10 minute dive, not a 2 hour dive.
[image]

closed 2020-12-13 by dirkhh

satoschu, 2020-12-13:
Hello Dirk, thank you for the description of the functionality of subsurface. Have a nice evening!