subsurface/subsurface #3095: 4.9.10 not working on macos 10.12.6

Ospeleo, 2020-11-30:
### Issue long description:
Since I installed 4.9.10 on my old macos 10.12.16 laptop, Subsurface won't start.

(missing library?)
It was working with 4.8.6
The subsurface website says that 4.9.10 should work on MacOS 10.10+

I'm not expecting a bug correction on such an old system, I'm just reporting the bug.

### Operating system:

OS Version:            Mac OS X 10.12.6 (16G2136)

Model: MacBookPro7,1, BootROM 68.0.0.0.0, 2 processors, Intel Core 2 Duo, 2,4 GHz, 8 GB, SMC 1.62f7
Graphics: NVIDIA GeForce 320M, NVIDIA GeForce 320M, PCI, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1067 MHz, 0x859B, 0x435434473353313036374D2E593136460000
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1067 MHz, 0x859B, 0x435434473353313036374D2E593136460000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 5.0.5f7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: ST1000LM048-2E7172, 1 TB
Serial ATA Device: Samsung SSD 850 EVO 1TB, 1 TB
USB Device: USB Bus
USB Device: USB Bus
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: Apple Internal Keyboard / Trackpad
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: USB 2.0 Bus
USB Device: Card Reader
Thunderbolt Bus: 

### Bug report

Process:               Subsurface [650]
Path:                  /Applications/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               4.9.10 (4.9.10)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Subsurface [650]
User ID:               501

Date/Time:             2020-11-29 22:29:02.078 -0500

Report Version:        12
Anonymous UUID:        20D0B810-2181-AC94-AA8D-D3D9B3FC4D4D

Time Awake Since Boot: 360 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Symbol not found: _OBJC_CLASS_$_MTLCaptureManager
  Referenced from: /Applications/Subsurface.app/Contents/MacOS/../Frameworks/QtGui.framework/Versions/5/QtGui (which was built for Mac OS X 10.13)
  Expected in: /System/Library/Frameworks/Metal.framework/Versions/A/Metal
 in /Applications/Subsurface.app/Contents/MacOS/../Frameworks/QtGui.framework/Versions/5/QtGui
[... 212 lines cut ...]
    0x7fff9d031000 -     0x7fff9d036ff3  libmacho.dylib (898) <17D5D855-F6C3-3B04-B680-E9BF02EF8AED> /usr/lib/system/libmacho.dylib
    0x7fff9d037000 -     0x7fff9d039ff3  libquarantine.dylib (85.50.1) <12448CC2-378E-35F3-BE33-9DC395A5B970> /usr/lib/system/libquarantine.dylib
    0x7fff9d03a000 -     0x7fff9d03bffb  libremovefile.dylib (45) <38D4CB9C-10CD-30D3-8B7B-A515EC75FE85> /usr/lib/system/libremovefile.dylib
    0x7fff9d03c000 -     0x7fff9d054ff7  libsystem_asl.dylib (349.50.5) <096E4228-3B7C-30A6-8B13-EC909A64499A> /usr/lib/system/libsystem_asl.dylib
    0x7fff9d055000 -     0x7fff9d055ff7  libsystem_blocks.dylib (67) <10DC5404-73AB-35B3-A277-A8AFECB476EB> /usr/lib/system/libsystem_blocks.dylib
    0x7fff9d056000 -     0x7fff9d0e3fef  libsystem_c.dylib (1158.50.2) <E5AE5244-7D0C-36AC-8BB6-C7AE7EA52A4B> /usr/lib/system/libsystem_c.dylib
    0x7fff9d0e4000 -     0x7fff9d0e7ffb  libsystem_configuration.dylib (888.60.3) <094DBBF4-276A-3A11-8AF3-72743CC338E6> /usr/lib/system/libsystem_configuration.dylib
    0x7fff9d0e8000 -     0x7fff9d0ebfff  libsystem_coreservices.dylib (41.4) <7D26DE79-B424-3450-85E1-F7FAB32714AB> /usr/lib/system/libsystem_coreservices.dylib
    0x7fff9d0ec000 -     0x7fff9d104fff  libsystem_coretls.dylib (121.50.4) <EC6FCF07-DCFB-3A03-9CC9-6DD3709974C6> /usr/lib/system/libsystem_coretls.dylib
    0x7fff9d105000 -     0x7fff9d10bfff  libsystem_dnssd.dylib (765.50.11) <B87F96B8-2694-3DB3-9923-1B4E2245C8BF> /usr/lib/system/libsystem_dnssd.dylib
    0x7fff9d10c000 -     0x7fff9d135ff7  libsystem_info.dylib (503.50.4) <611DB84C-BF70-3F92-8702-B9F28A900920> /usr/lib/system/libsystem_info.dylib
    0x7fff9d136000 -     0x7fff9d158ff7  libsystem_kernel.dylib (3789.73.50) <932ACBB6-9962-3839-AEE3-D8AA6BF1DD02> /usr/lib/system/libsystem_kernel.dylib
    0x7fff9d159000 -     0x7fff9d1a0fe7  libsystem_m.dylib (3121.6) <86D499B5-BBDC-3D3B-8A4E-97AE8E6672A4> /usr/lib/system/libsystem_m.dylib
    0x7fff9d1a1000 -     0x7fff9d1bfff7  libsystem_malloc.dylib (116.50.9) <82647590-DAEF-3499-8BF3-F1E3048861FE> /usr/lib/system/libsystem_malloc.dylib
    0x7fff9d1c0000 -     0x7fff9d219ffb  libsystem_network.dylib (856.60.1) <369D0221-56CA-3C3E-9EDE-94B41CAE77B7> /usr/lib/system/libsystem_network.dylib
    0x7fff9d21a000 -     0x7fff9d223ff3  libsystem_networkextension.dylib (563.60.2) <B021F2B3-8A75-3633-ABB0-FC012B8E9B0C> /usr/lib/system/libsystem_networkextension.dylib
    0x7fff9d224000 -     0x7fff9d22dff3  libsystem_notify.dylib (165.20.1) <B8160190-A069-3B3A-BDF6-2AA408221FAE> /usr/lib/system/libsystem_notify.dylib
    0x7fff9d22e000 -     0x7fff9d236fe7  libsystem_platform.dylib (126.50.8) <897462FD-B318-321B-A554-E61982630F7E> /usr/lib/system/libsystem_platform.dylib
    0x7fff9d237000 -     0x7fff9d241ff7  libsystem_pthread.dylib (218.60.3) <B8FB5E20-3295-39E2-B5EB-B464D1D4B104> /usr/lib/system/libsystem_pthread.dylib
    0x7fff9d242000 -     0x7fff9d245ff7  libsystem_sandbox.dylib (592.70.2) <19320A42-2E3B-361B-BBDA-2F5F2E87B100> /usr/lib/system/libsystem_sandbox.dylib
    0x7fff9d246000 -     0x7fff9d247ff3  libsystem_secinit.dylib (24.50.4) <F78B847B-3565-3E4B-98A6-F7AD40392E2D> /usr/lib/system/libsystem_secinit.dylib
    0x7fff9d248000 -     0x7fff9d24fffb  libsystem_symptoms.dylib (532.50.48) <1C822669-BF39-3639-A00B-B9170A63F342> /usr/lib/system/libsystem_symptoms.dylib
    0x7fff9d250000 -     0x7fff9d263ff7  libsystem_trace.dylib (518.70.1) <AC63A7FE-50D9-3A30-96E6-F6B7FF16E465> /usr/lib/system/libsystem_trace.dylib
    0x7fff9d264000 -     0x7fff9d269ffb  libunwind.dylib (35.3) <3D50D8A8-C460-334D-A519-2DA841102C6B> /usr/lib/system/libunwind.dylib
    0x7fff9d26a000 -     0x7fff9d293ff7  libxpc.dylib (972.70.5) <255AA6A1-26AA-33F6-BC11-3DCE12A0BC0F> /usr/lib/system/libxpc.dylib

dirkhh, 2020-11-30:
I haven't made a macOS 10.12 version of 4.9.10
You can get 4.9.8 here: https://subsurface-divelog.org/downloads/Subsurface-4.9.8-10.12.dmg
I'll try to find time today to make a 4.9.10 version as well - will be in the same place, similar naming pattern.

(we have something like 5 users that are on macOS 10.12, based on server logs - it's kinda hard to justify the effort)

closed 2020-11-30 by dirkhh

dirkhh, 2020-12-01:
And here's a macOS 10.12 version of 4.9.10

https://subsurface-divelog.org/downloads/Subsurface-4.9.10-10.12.dmg