subsurface/subsurface #3141: Release CI snaps to the Snap Store

Saviq, 2021-01-08:
### Describe the issue:

- [x] Discussion request

### Issue long description:
We've had very good experience with snaps built out of PRs getting published on [the Snap Store](https://snapcraft.io/subsurface/). It allows for a simple `snap refresh subsurface --channel edge/pr1234` to get to a snap built out of a pull request. With `snap save`/`snap restore` you can ensure that your data is safe, too.

The problem is that credentials are required to upload to the store, and those are obviously sensitive. GitHub doesn't allow access to secrets from external pull requests, as malicious PRs could steal those.

Depending on how Subsurface development happens, the simplest option is to _only_ publish to the store from internal PRs (those coming from the `subsurface/subsurface` repository).

We've been using [bors](https://bors.tech/) to help with external contributions, as a `bors try` comment from an authorized user would trigger a full run that has access to secrets after they confirmed the code is sane.

Another option (that complements the above) for regular/trusted developers would be to create a team that was privileged and run a workflow on the [`pull_request_target` event](https://docs.github.com/en/free-pro-team@latest/actions/reference/events-that-trigger-workflows#pull_request_target) for the sensitive bits, that would only run if the actor was a member of that privileged team.

WDYT?

### Operating system:
Linux

### Subsurface version:
N/A

### Mentions:
@dirkhh

dirkhh, 2021-01-08:
So we have tried for a while to create binaries for PRs but the download numbers were basically zero. The community that we have very very rarely seems to be open to testing versions based on a PR (we often struggle to get people to test, provide feedback on things that are in the main branch already). As a result we have completely stopped making binaries for PRs available (and if people actually want to test things and can't build, I do the occasional one-off manually).
That's why I suggested to create Snaps (and upload then) after a PR has been merged, so for pushes to master.
And that solves the problem with secrets quite nicely.

Saviq, 2021-01-09:
OK then, pushes to master are published on the `edge` channel already. For PRs, they're available as an artifact on the workflow (you may wanna check in settings what's your retention on those to avoid running out of GitHub quota).

I'll close this, then :)

closed 2021-01-09 by Saviq