subsurface/subsurface #3143: build.sh behavior

salvadorcunat, 2021-01-09:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [X] Discussion request

### Issue long description:

Line

`490 if [ "$QUICK" != "1" ] && [ "$BUILD_DESKTOP$BUILD_MOBILE" != "" ] ; then`

in build.sh will always fail if the script is not called with -desktop, -mobile or -both flags, making, in fact, -quick to be the default, and thus calling

`subsurface/scripts/build.sh` 

acts as

`subsurface/scripts/build.sh -quick -desktop`

This way, in a fresh install, only libdc and desktop subsurface are built (no googlemaps plugin).

Not a big issue as one can always call -desktop, -both or -all to get the plugin in place.
It's simply I don't know if it's the desired behavior as it's not documented neither in the script nor in the INSTALL file.

BTW 6a7152b only talks about not building googlemaps in "downloader only" builds, mainly useful on raspberry pi devices.  Setting a default value BUILD_DESKTOP=1, and unseting it in -mobile and -downloader options would give a better approach, building every thing unless -quick given.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2021-01-09:
So the logic of `-quick` is that if you set it the dependencies aren't rebuilt. And that includes `googlemaps` (which actually takes fairly long to build and never ever changes and only needs to be rebuilt if you change Qt version, TBH.
So if I think through the logic of this if statement, it seems to say to only build `googlemaps` if `-quick` wasn't set and you are building either desktop or mobile. Which to me seems right. But then again, I may be completely missing your point, so please humor me and explain why this is the wrong thing to do?

salvadorcunat, 2021-01-09:
Hi Dirk.
It seems I explained myself quite bad, as usual.
I'm not telling this is wrong (as said, I'm fine with it). What I was trying to explain is that the script behaves just opposed as it used to.
Till now, if one wanted to do a fresh build, one just had to blow away install-root and build dir, and call build.sh without flags. This brought in place every needed thing.
Currently using flag -desktop, -both or -all, without -quick is mandatory.
Thus, I can only see two ways to manage the change:
- make the script work like it used to
- document the change in INSTALL or, at least, in the script header, and keep going.

The background of the history is that installing libqt5charts5 triggered a big upgrade in my debian sid distro, and after it, I've spent 1/2 hour trying to figure out why googlemaps wasn't build as it used to be. ;-)

dirkhh, 2021-01-09:
Thank you for your persistence. I now understand what you are saying (I think).
When called with no argument at all, the script now behaves differently from before.
I think this is a valid fix?
```
diff --git a/scripts/build.sh b/scripts/build.sh
index 1c87275ae..52e30c0c9 100755
--- a/scripts/build.sh
+++ b/scripts/build.sh
@@ -133,6 +133,11 @@ while [[ $# -gt 0 ]] ; do
        shift
 done
 
+# recreate the old default behavior - no flag set implies build desktop
+if [ "$BUILD_MOBILE$BUILD_DOWNLOADER" = "" ] ; then
+       BUILD_DESKTOP="1"
+fi
+
 if [ "$BUILD_DEPS" = "1" ] && [ "$QUICK" = "1" ] ; then
        echo "Conflicting options; cannot request combine -build-deps and -quick"
        exit 1;
```

salvadorcunat, 2021-01-09:
Yes, It will work.
Sorry, sometimes I tend to depict the forest instead of depict the tree.
Thanks Dirk.

PR #3146 (2021-01-10): build.sh: recreate previous behavior with no args
When calling build.sh with no args asking for a specific build type, that
should be equivalent with calling it with the -desktop arg.

Reported-by: Salvador Cuñat <salvador.cunat@gmail.com>

### Describe the pull request:

- [x] Bug fix

### Pull request long description:

restore the previous behavior by ensuring at least one of the `BUILD_XXX` variables is set, and if none are set, assume `BUILD_DESKTOP`

### Related issues:

Fixes #3143 

### Mentions:
closed 2021-01-10 by dirkhh