subsurface/subsurface #3144: Initial time of manual added dive is ahead by 1 hour

PeterZaal, 2021-01-09:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I am currently in the process of adding the history of a lot of old dives to my log. For a lot of these dives I only have them on paper, so I am manually adding these dives. I noticed that when adding a dive manually the initial date and time are set at the current date and time. At least that is what it is supposed to happen I think, but the time is ahead by 1 hour.
This is indepedent of the Windows timezone setting; whatever timezone is selected, the initial time is always ahead by 1 hour comparing to the current time.
Ofcourse this is not a big issue, since this time will be edited to represent the time of the dive you are adding. But it was just something I noticed.

### Operating system:
Windows 10

### Subsurface version:
Official 4.9.10

### Steps to reproduce:
Press Ctrl+ + to add a dive
Notice that the time on the Notes tab is ahead by 1 hour

### Current behavior:
Initial time for a manually added dive is ahead by 1 hour

### Expected behavior:
Initial time for a manually added dive is equal to the current time

dirkhh, 2021-01-09:
So this will seem silly to some, but when we first implemented this the idea was "well, the person entering the dive isn't diving right now - so we could either assume that they just came back from a dive, or that they are just about to go." And I don't recall why we ended up with the second option and said "yeah, let's assume the dive is in an hour".

So I __think__ that you'll see that the dive time stored is correct (we had quite a while where it was indeed wrong because of a UTC vs local time bug, and for you in the NL it would indeed have been wrong by an hour in the winter, but I really hope that this is long fixed).

Can you verify that this is a correct statement?

PeterZaal, 2021-01-09:
Yes the dive time is absolutely stored correctly, no issues with that. That's also why I tried different time zones.
I think the reasoning "the person is about to go diving" seems a strange; why would you add a dive you are about to make but don't yet have any data for? But since it's intentionally not at the current time, and it's really a non-issue since you will enter the correct time anyway, I will close.
Thanks for the explanation!

closed 2021-01-09 by PeterZaal