subsurface/subsurface #3149: O2 % for Cylinder with 21% is not saved / shown

PeterZaal, 2021-01-10:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I have some dives where I did enter 21% for the O2 % field in a row in the Cylinder grid. When reopening the dive log from cloud storage (this is after some time) the 21% is missing, it's empty. Also, when I save the log to a local file there is initially no o2 attribute in the xml. But even than after re-entering 21% and saving locally, I can see the o2='21.0%' in the xml, but after restarting Subsurface and reopening this local logbook the O2 percentage shows empty.
I can imagine that 21% is ofcourse some kind of default, but to me there is a big difference between empty meaning 'I dont know the percentage' or meaning 'it's the default, air'.
If I know I dived with air I enter 21%, if I don't know or don't remember any more what it was I leave it empty. An entered value should always be left as-is anyway and not be changed or removed.

### Operating system:
Windows 10

### Subsurface version:
Official 4.9.10

### Steps to reproduce:
- In equipment tab use a Cylinder and enter 21 in the O2 % field.
- Save changes and close Subsurface
- Restart Subsurface and (if needed) reopen the logbook
- Notice that the O2 field is empty

### Current behavior:
An entered value of 21% for Cylinder O2 % is not shown or removed

### Expected behavior:
An entered Cylinder O2 % value should always be remained

dirkhh, 2021-01-10:
This is a long ongoing debate.
Linus is adamant that air is special and that air needs to be stored as 0/0.
Feel free to argue with him :-)

PeterZaal, 2021-01-10:
Well, let's try :), can I ping Linus here? @torvalds does that work?

I have the strong opinion that if I enter a value (whatever it is) that should be preserved. Maybe formatted in the correct format, but the value should be preserved. Subsurface should not remove my entered value.
It's a bit like with SQL, a difference between NULL and '' (empty string), this is very different. One is 'unknown', the other is literally empty string.
21% is not special, it's 21% like any other percentage. To be honest, for me it is kind of special, because I almost never dive with air. So IF I dive with air, I want this to be recorded as-is. But that is for any value I enter.
And, it seems like it is not stored as 0/0, but the entered value is removed (because if you save to a local logbook the o2 attribute is not there) (but maybe internally that is the same as 0/0). To me this is really weird that an entered value is changed or removed, and makes it impossible to distinguish between air and 'i don't remember the percentage'.

PeterZaal, 2021-01-10:
It's also not consistent with the statistics. I just tried with a single dive with no O2 %. The statistics shows O2 (max): 20.9%.

dirkhh, 2021-01-10:
Linus has GitHub notifications turned off (for rather obvious reasons). I'll ping him offline.

torvalds, 2021-01-10:
> I have the strong opinion that if I enter a value (whatever it is) that should be preserved. Maybe formatted in the correct format

That's exactly what it does.

It preserves the value, and then formats it in the correct format.

The correct format for "air" just happens to be empty.

And no, there is no difference between 20.9 and 21%

Seriously.  Somebody who thinks 20.9% is different from 21% is just barking up the wrong tree.

And no, you cannot have a breathable mixture with 0 air, so that's not relevant either.

And that is *my* strong opinion. I don't want to see some idiotic 20.9 or 21 percent garbage.

torvalds, 2021-01-10:
> The correct format for "air" just happens to be empty.

One more note: this is entirely consistent with all our other formatting. We don't show nitrox as 32/0 to show that there is zero Helium. Because we don't show the pointless standard case. So the formatting is entirely consistent, and entirely unambiguous.

And no, we're not some kind of SQL database, but if that is your world-view, then that is simple to explain too. Cylinders have a gas mix. End of story. In SQL terms, it's a "NOT NULL" constraint, and you cannot add a cylinder without adding a value.

Honestly, it was probably a mistake to allow fractional percentage points in the gas mix at all, so the 20.9% vs 21% issue is a bit unfortunate, but then turning them both into just "air" is the solution to _that_ ambiguity.

In the defense of the extra precision existing at all - while most data loggers (ie dive computers) don't even allow that in the first place, that otherwise pointless precision had its own reasons too (ie "import from other sources"), so it's understandable even if it's not wonderful.

I have never seen a scuba gas analyzer that would be remotely precise enough that the decimal digit would actually _matter_ (they probably exist, although not in any dive shop I've ever seen - presumably medical grade stuff), but people do often want to log the exact value that their analyzer told them even if it's false precision, so I guess that's another reason to allow that tenth-of-a-percent thing.

And I guess you could use that pointless extra precision - if you want to - and decide to use "21.1%" as a "this is _explicitly_ air, and work around things that way.

Final note: when I actually test this, I notice that we seem to have screwed up the "20.9 and 21 is air" at data entry time. So that's a bug. If you enter 21% in the equipment tag, it actually will show up as 21% in the dive list. Until you save and reload it, which will fix it.

I'm not sure when that bug crept in. But we do have a bug here.

torvalds, 2021-01-10:
> Final note: when I actually test this, I notice that we seem to have screwed up the "20.9 and 21 is air" at data entry time. 

Well, that was a trivial one-liner and should be fixed by #3151

I didn't do extensive testing, but the patch is trivial and the dumb test-case I checked was good.

Adding a note to @bstoeger here, since my one-liner touches his model code. I _think_ it's ObviouslyCorrect(tm), but no harm in double-checking.

PeterZaal, 2021-01-10:
Thanks for the explanation Linus.
And no, this is absolutely not about a difference between 20.9% or 21%. I understand how that works ;) I usually dive with a 10/70 mix and in this case it actually might matter a little if it's closer to 9 or 11%, but this was and is not about a 0.1 difference.
That just came by because I was curious how an empty value would be seen by the new statistics module.

The analogy with SQL I was trying to make was that there is a difference between two values, although on the screen (with formatting) they look the same. Ofcourse there is the same problem with editing: should an empty value be saved as 'unknown' or 'empty' (or 'default'). It's one or the other, afterwards you cannot distinguish anymore.

So if the decision has been made that empty is de display value for air, that's what it is. All other kinds of decisions could have been made, like adding a default value of 21 when adding a cylinder, but that is not what happened and I completely understand that this is not something one would change.
Maybe it's because I hardly ever dive with air anymore this stood out as strange for me and it does make it impossible to have an 'unknown' gas mix. Even if I dive with air I will analyse it and label it with the percentage, so I personally don't use empty values. A tank with no label on (= empty value) it is for me an 'unknown gas mix'.
And yes I understand in reality there is alway a gas mix, and having an unknown gas mix probably brings many other issues, like not being able to calculate deco etc.

I can at least understand a bit why it is how it is. Thanks again for explaining your point of view.

closed 2021-01-10 by dirkhh