subsurface/subsurface #3150: Weight system not saved in cloud storage

PeterZaal, 2021-01-10:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
So this is really weird. I have some older dives where I know for sure I entered the weight system, type and weight. When I now look at these dives the weight type is missing, it's empty. Even when I select again a type, save the change to cloud, it is not preserved. After restarting Subsurface it's empty again!
This also happens with trying to change a weight (type or kg), or adding a weight system, it is not preserved. After restarting Subsurface the old value is back (or not preserved at all).
This only seems to happen when saving and retrieving data from cloud storage. When I use a local logbook the type is saved and preserved after restarting.
This is really annoying because it seems like a lot of my dives of which I am 100% sure I entered the weight type (literally 100's and 100's of dives) are now missing the weight type.

### Operating system:
Windows 10

### Subsurface version:
Official 4.9.10

### Steps to reproduce:
- For a dive with an existing weight system, change the weight type
- Save the change to the cloud storage and close Subsurface
- Restart Subsurface and notice the weigth type of that dive is unchanged

### Current behavior:
Weight system is not preserved on saving to the cloud storage

### Expected behavior:
Weight system should be saved in the cloud storage

### Additional information:

### Mentions:

dirkhh, 2021-01-11:
Sorry to not respond to that earlier. That is of course rather bad.
If the data was ever stored in the cloud, we should be able to get it back (potentially a bit painful, but doable). But of course if you entered the data with a broken version and it was never stored, that's bad.

Some testing seems to indicate that what is happening is slightly different from what you report.
At least when testing with current master.

It's still BAD.

So if the ONLY change that has been made since the last save of the data was a change to weight systems and the user saves changes, the dive list is marked 'unchanged' (i.e., everything is saved to disk), but git storage isn't actually updated.

But if you make another, unrelated change to the same dive (e.g., edit the notes field), then both changes as stored.

torvalds, 2021-01-11:
Sent an email, but adding @bstoeger here on github too.

It sounds very much like a missing "invalidate_dive_cache()": we change the dive data, but never tell the git save logic that "this dive no longer matches what you loaded".

Right now we do that "invalidate_dive_cache()" almost randomly for different parts of the dive. I _think_ we should strive to do it whenever we start editing a dive (even if the edits are then undone) so that there isn't this kind of "every small detail needs to know about this invalidation". 

Berthold?

PeterZaal, 2021-01-11:
Yes your observation is correct Dirk, if I change another field everything is saved.
I actually noticed it because the only thing I wanted to change on (a lot of) old dives was the type of weight system.
To me it seems this data (the weight system type, not the amount) has been lost at some point, but I cannot tell when that happened. This can be more than 2 years ago because currently I don't use any weight anymore for at least this period, so it only shows when looking at older dives. Although... I have one trip in Sep 2019 to Bonaire where I use 3 kg weight, and all these dives now have either 5.0 or 5.5 kg with them. That is strange because I am sure I entered 3 kg, I have never used 5 or 5.5 kg. Ofcourse I could have made a mistake in one of the dives, but not all 26. These dives are imported from my Shearwater and afterwards I added the weight system, but also entered all the other info.
But I also have trips from 2016 where both the type and weight are correct, and another trip from 2017 where the type is empty and the weight correct.
So to me it's all strange and not consistent, I cannot see any relation to dives that are correct and those that have lost or incorrect information.

bstoeger, 2021-01-11:
> It sounds very much like a missing "invalidate_dive_cache()": we change the dive data, but never tell the git save logic that "this dive no longer matches what you loaded".

Yes, it is missing the call in the undo() and redo() member functions ("methods") of the respective undo commands (compare to the cylinder undo commands). Note that the weight undo commands are not based on EditBase, but EditWeightBase. I will submit a PR before lunch (local time).

> I think we should strive to do it whenever we start editing a dive (even if the edits are then undone) so that there isn't this kind of "every small detail needs to know about this invalidation".

Not sure I agree. Because think about the situation 1) edit, 2) save, 3) undo and variations thereof. I really think this should be done with a high granularity. For the common edit commands this works quite well with the EditBase class, which calls into virtual functions of the base class. Perhaps I should have followed the pattern for the other Edit* base classes.

The plan I had for a long time is to only store and pass around "const struct dive *" and these have to be converted to a "struct dive *" using a "get_mutable_dive(d)" call, which invalidates the dive cache. One could even make this explicit by typedefing these to dive_ptr and mutable_dive_ptr, respectively. However, that means making the application "const-correct" and so far I lacked the energy to do so.

commit 5692f0d6 (2021-01-11): undo: add missing invalidate_dive_cache() calls

The AddWeight, RemoveWeight, EditWeight and ReplanDive
commands were missing invalidate_dive_cache() calls.
Add them to ensure that the dives are written to git
logs on save.

Fixes #3150

Reported-by: Peter Zaal <peter.zaal@gmail.com>
closed 2021-01-11 by dirkhh