subsurface/subsurface #3182: Feature request: CCR setpoint changes (default & manual) during calculated ascents

justin-ashworth-research, 2021-01-23:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

When editing a CCR dive in the planner, we would like for setpoint changes to be included (or a parameter) in the auto-magically generated ascent plan. 

For example, it is typical to adjust our ppO2 setpoint up to 1.4 during intermediate decompression stops, and up to 1.6 during the 6 metre decompression stop (i.e. oxygen rebreather mode). This common practice dramatically accelerates decompression and decreases runtime.

I have not yet examined the codebase to decide whether it would be appropriate for me to attempt a pull request.

### Operating system:

### Subsurface version:

4.9.10

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2021-01-25:
I understand this is a feature that could be useful. As you might have noticed, the user interface of the planner is already quite (some say far too) complex. How would you imagine to configure such set point change from a UI perspective? Would that happen at a specific depth/stop or how are the details determined. Please take into account that such a feature should not only scratch your personal itch but if implemented should help many users. Could you reasonably expect others to have different strategies here? How can we make most of the people happy without adding a million more configurable options that will confuse the rest of our users?

justin-ashworth-research, 2021-01-28:
I made a fork out of interest and added a function that seems to accomplish this in core/planner.c:
https://github.com/justin-ashworth-research/subsurface/commit/b5422eb7cf1941d8dba682a35f17b0fd7b8d926b

However I have not attempted to make this flexible within the interface.

justin-ashworth-research, 2021-01-28:
I put the setpoint updates too late in plan() in the above commit--too clean! (and not updating the deco)

justin-ashworth-research, 2021-01-28:
This seems to accelerate the deco... akin to what we aim for on some tech CCR dives:
https://github.com/justin-ashworth-research/subsurface/commit/4e3f31c99df3fe7394569bc83d6b9191d87d5d20

The plan() function is quite verbose and monolithic! Impressive work

atdotde, 2021-01-28:
As I said: the challenge is not to incorporate this in the planning but to come up with a decent user interface.

justin-ashworth-research, 2021-01-28:
Yes, perhaps a single checkbox. In the CCR tech deco courses it is typically taught as setpoint to 1.4 at 21 metres, and to 1.6 at 6 metres (O2 flush/O2 rebreather mode). 
Or more fields in the "tech setup" panel of the options. At the least, the magic (arbitrary) numbers probably belong in the "prefs" global (?)

atdotde, 2021-01-28:
I see at least four numbers to be configured (two depths and two set-points). And why stop at two pairs? Yes, you can try to hide them in the preferences. But I like to think of the unique selling point of our planner that all changes have immediate responses in the dive plan so you can interactively see what you are doing. This is not really accessible when stuff is not in the main panel (which is overloaded) but in a separate preferences window.

Please convince me that this model of (depth dependent) set-point changes is something that many CCR divers do (I am not one myself) and that all these people will be catered with two pairs of depth/set-point pairs. 

If we add stuff to the UI, it only makes sense if it is useful to a significant part of the user base and not 90% of the user's reaction is "nice, but I would like to have it slightly different".

justin-ashworth-research, 2021-01-28:
Here I put in a checkbox to toggle the setpoint increases while in CCR mode:
https://github.com/justin-ashworth-research/subsurface/commit/321a5150ceccf637e979a55d190318e6f3c8f729

These CCR setpoint increases during decompression are taught in the advanced deco CCR courses, e.g. through TDI and IANTD.

Probably at some point other deco CCR users will be looking for this capability.

In other programs (Multideco?), the way it is accomplished is that the user adds a special "change setpoint" item to the plan, at the specified depth. This would be different/independent from a dive plan "segment" in Subsurface, and it would need to interact nicely with the iteratively auto-generated deco segments. Essentially I think this is what I have implemented in the fork, except as a checkbox rather than a manually entered "step" in the dive plan.

PeterZaal, 2021-01-29:
On Thu, Jan 28, 2021 at 6:58 PM justin-ashworth-research <
notifications@github.com> wrote:

> Here I put in a checkbox to toggle the setpoint increases while in CCR
> mode:
> justin-ashworth-research@321a515
> <https://github.com/justin-ashworth-research/subsurface/commit/321a5150ceccf637e979a55d190318e6f3c8f729>
>
> These CCR setpoint increases during decompression are taught in the
> advanced deco CCR courses, e.g. through TDI and IANTD.
>
> Probably at some point other deco CCR users will be looking for this
> capability.
>
> In other programs (Multideco?), the way it is accomplished is that the
> user adds a special "change setpoint" item to the plan, at the specified
> depth. This would be different/independent from a dive plan "segment" in
> Subsurface, and it would need to interact nicely with the iteratively
> auto-generated deco segments. Essentially I think this is what I have
> implemented in the fork, except as a checkbox rather than a manually
> entered "step" in the dive plan.
>
Yes, changing (to a higher) setpoint during deco is quite common for CCR
divers.
This is how that looks in Multideco:
[image: Screenshot_20210129-133522.jpg][image:
Screenshot_20210129-133530.jpg]

cyberdiamond, 2021-01-29:
I have not built subsurface with this patch yet to look at it but just some thoughts based on the discussion.

Background: I am an Advanced Hypoxic Trimix CCR diver.

I dive a manual ccr so don't actually have set point changes but in practice I do bump up the ppo2 to at least 1.4 (and higher as long as I am not working hard) during any deco stops.

I don't do a strict 1.4 at 21m for instance I might bump it up to approx 1.5 / 1.6 at the start of the stop and let it fall to say 1.3 by the end before moving up to the next stop while releasing gas and adding o2 to approx 1.5 / 1.6 again and repeating so for planning the 1.4 would work as an average. 

I would not agree with a 1.6 ppo2 auto setting though (1.5 might be more appropriate).
Reason being is that at 6m it is not really possible to get to or maintain a ppo2 of 1.6 when on a rebreather.
Even if I unplug the diluent and thereby turning it into an oxygen rebreather, empty the counter lung and flush with pure o2.
The first breath out of my lungs includes all the nitrogen (and helium if using trimix) that is coming out of my body and quickly dilutes the ppo2 in the ccr to 1.5 and lower the longer you breath and yes when it gets down a bit I again flush pure o2 it to bring it back up as high as possible.

So that max ppo2 should be settable instead of 1.6 which is less conservative than reality which is never a good thing when planning but overall I agree that it would be a useful feature.

Steve

cyberdiamond, 2021-01-29:
> On Thu, Jan 28, 2021 at 6:58 PM justin-ashworth-research < ***@***.***> wrote: Here I put in a checkbox to toggle the setpoint increases while in CCR mode: ***@***.*** <[justin-ashworth-research@321a515](https://github.com/justin-ashworth-research/subsurface/commit/321a5150ceccf637e979a55d190318e6f3c8f729)> These CCR setpoint increases during decompression are taught in the advanced deco CCR courses, e.g. through TDI and IANTD. Probably at some point other deco CCR users will be looking for this capability. In other programs (Multideco?), the way it is accomplished is that the user adds a special "change setpoint" item to the plan, at the specified depth. This would be different/independent from a dive plan "segment" in Subsurface, and it would need to interact nicely with the iteratively auto-generated deco segments. Essentially I think this is what I have implemented in the fork, except as a checkbox rather than a manually entered "step" in the dive plan.
> Yes, changing (to a higher) setpoint during deco is quite common for CCR divers. This is how that looks in Multideco: [image: Screenshot_20210129-133522.jpg][image: Screenshot_20210129-133530.jpg]

P.S. Can't see your screenshots.

PeterZaal, 2021-01-30:
Ah I replied on the mailinglist, that was not so clever, images get lost here. Attached to this comment.

And yes I do about the same procedure, also manual CCR (KISS). Depending on depth and length of stops sometimes start bumping the PO2 sooner than 21m. At 6m I am already quite happy if I get 1.5.
But for planning purposes I usually put in 1.4 for an average.

[image]
[image]

atdotde, 2021-01-30:
OK, you guys convinced me that this could be useful. Contemplating this a bit, I realized that such a depth dependent set point change is probably closest in spirit to a gas change (as you would do it as OC diver). And as long as you are not planning bailout, the gas lists is also at most semi-useful in CCR mode. So maybe it would make sense (and would cause minimal UI distortion) to implement this via the gas list: Have fake cylinders associated to a set point that the planner would switch to a certain depths. What do you think?

justin-ashworth-research, 2021-01-30:
I am curious to see what can work ideally within the implementation (e.g. core/planner.c?), with which I am quite unfamiliar! I only got as far as iteratively and manually resetting the po2 variable in plan().

atdotde, 2021-01-30:
Currently, gas changes are ignored in the planner if in CCR mode. We could turn those on for example if the cylinder (that would be the cheapest way) had a name like "SP 1.4" and interpret those as set-point changes. I guess, you don't have to do a lot to the code that handles gas changes to make that possible. 

Slightly more ambitions would be to add a column to the gas model which holds the set-point values. Do you want to give that a try?

justin-ashworth-research, 2021-01-31:
Ok I will try to understand further what is happening in the code, so that I can see how this might be done.

justin-ashworth-research, 2021-02-04:
Small note, I discovered a small issue whereby logic in the trunk omits CNS and OTU accrual during deco stops while in CCR:bailout mode. I think I fixed that in the fork.

justin-ashworth-research, 2021-02-05:
An intuitive thing would be to simply add "timeless" elements to the "dive planner points" panel/table itself: depth-dependent setpoint shifts, maintained in the order added by the user, with no duration or runtime given (perhaps "-1" for one or both time fields). These could then be parsed as setpoint shifts when ascent segments cross their depth thresholds.

However, numerous files and process loops in the code are resistant to using "-1" as a pseudo-time, or supporting the entry of pseudo-stops without messing up other assumptions, ordering and times of other dive points. I noticed and experimented with several places where this can be attacked, but hunting around for everywhere that needs to change has exhausted me for the moment.

atdotde, 2021-02-21:
Justin,

> On 5. Feb 2021, at 08:00, justin-ashworth-research <notifications@github.com> wrote:
> 
> An intuitive thing would be to simply add "timeless" elements to the "dive planner points" panel/table itself: depth-dependent setpoint shifts, maintained in the order added by the user, with no duration or runtime given (perhaps "-1" for one or both time fields). These could then be parsed as setpoint shifts when ascent segments cross their depth thresholds.
> 
> However, numerous files and process loops in the code are resistant to using "-1" as a pseudo-time, or supporting the entry of pseudo-stops without messing up other assumptions, ordering and times of other dive points. I noticed and experimented with several places where this can be attacked, but hunting around for everywhere that needs to change has exhausted me for the moment.
> 
> 

finally, I found some time to implement this, please see

https://github.com/subsurface/subsurface/pull/3202 <https://github.com/subsurface/subsurface/pull/3202>

I am happy with the logic of treating this like a gas change but the user interface (add a cylinder with the name „SP 1.4“) is a hack and needs a better idea. Please check that this is what you intended.

Best
Robert

closed 2021-03-05 by atdotde