subsurface/subsurface #3184: Profile shows unexpected bearing if cursor is after the end

PeterZaal, 2021-01-25:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Just a very small thing I noticed. See attached image: when the cursor is placed somewhere after the end of the dive, the profile shows an extra 'bearing: 0' line.

### Operating system:
Windows 10 Home

### Subsurface version:
4.9.10 official, 4.9.10-421 from test

[image]

torresjoj, 2021-09-07:
@dirkhh I think this is closed as of 7e6d993, right? Seems automation didn't catch it given you had to clean up the commits (sorry!) and the PR wasn't merged.

dirkhh, 2021-09-07:
Correct - my fault for not closing it.

closed 2021-09-07 by dirkhh

PeterZaal, 2021-10-18:
Just a short confirmation that I checked with latest official 5.0.4. and this is indeed resolved, thanks.