subsurface/subsurface #3192: Small bug in Android version of statistics tool

willemferguson, 2021-02-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When the statistics tool is active and one hits the Android BACK button (bottom right on my device), Subsurface terminates. I have no evidence that this is a crash. It is probably just an erroneous termination that has not been set to return to the dive list.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2021-06-28:
Is there a way to reproduce this on MobileOnDesktop? There is no back-button when on the statistics page for me.

dirkhh, 2021-06-28:
There's code in many places to 'catch' the back behavior... if we don't catch it, the app terminates on Android (it's literally the official way to exit an app). I need to check and see why this isn't on the statistics page. I thought I added this but we refactored the code quite a few times during its development :)

dirkhh, 2021-08-19:
BTW - Alt-left simulates the back button for me with Mobile-on-Desktop for Linux

commit 6cf5d766 (2021-08-19): mobile: don't quit on back button in Statistics mode

Fixes #3192
commit 503d8bd1 (2021-08-19): mobile: don't quit on back button in Statistics mode

Fixes #3192
closed 2021-08-19 by dirkhh