subsurface/subsurface #3193: Shearwater Peregrine is not visible in the Bluetooth devices overview

TTLManfred, 2021-02-09:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Searching for Bluetooth devices in subsurface desktop version doesn't show the Shearwater Peregrine. Connecting to mobile version of subsurface is possible.

### Operating system:
Win10 20H2, German
Lenovo V130
Shearwater Peregrine, Firmware v81 german

### Subsurface version:
4.9.10 (Desktop, German)

### Steps to reproduce:
Searching for Bluetooth device within subsurface "Import from dive computer"

### Current behavior:
Doesn't show the Peregrine

### Expected behavior:
Showing the Peregrine ;-)

### Additional information:

### Mentions:

dirkhh, 2021-02-09:
That's... surprising. 4.9.10 should support the Peregrine.
But this could be more of a Windows pairing issue than anything else.
Did you look at the user manual and the section about the oddities with Bluetooth Low Energy (BLE) devices and Windows?
The dive computer needs to first be paired with the OS before Subsurface can see it.

TTLManfred, 2021-02-09:
Yes, I've tried to pair it but it's not possible. Windows tries to connect with with the Peregrine. After some time, it stops the attempt without success.

closed 2021-08-19 by dirkhh