subsurface/subsurface #3195: XML data UDDF error when exporting from Susurface

UserSBSRF, 2021-02-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mturkia, 2021-02-12:
All the tags you mention are in the [UDDF 3.2.1](https://www.streit.cc/extern/uddf_v321/en/index.html) standard

UserSBSRF, 2021-03-21:
> All the tags you mention are in the [UDDF 3.2.1](https://www.streit.cc/extern/uddf_v321/en/index.html) standard

Hello,

Your answer is far too succinct.

You should read or reread the complete information about the uddf standard before responding.

It is not enough just to stack uddf standard tags in the export file.
There are controls to be respected on the mandatory presence of data in the tags.

In the future, be less assertive and more humble, you will gain credibility.

Best regards

atdotde, 2021-03-21:
> In the future, be less assertive and more humble, you will gain credibility.

I think this is great advice for everybody.

closed 2021-03-21 by atdotde