subsurface/subsurface #3203: Dive planner does not calculate CNS and OTU correctly for pSCR dives

arosl, 2021-02-25:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
While checking some profiles I noticed that the OTU and CNS are not calculated correctly for PSCR dives.
with settings: EAN36, 0.720ℓ/min, 1:6, 20 bottom sac. a 300min dive to 18m (p02 0.88) with oxygen at 6m should give the same output as a OC dive with EAN31. 

OC EAN31:
  | depth | duration | runtime | gas
-- | -- | -- | -- | --
➘ | 18m | 1min | 1min | EAN31
➙ | 18m | 299min | 300min |  
➚ | 6m | 1min | 301min |  
- | 6m | 34min | 335min | oxygen
➚ | 0m | 1min | 336min |  

CNS: 158%
OTU: 335

PSCR 1:6 EAN36:

  | depth | duration | runtime | gas
-- | -- | -- | -- | --
➘ | 18m | 1min | 1min | EAN36
➙ | 18m | 299min | 300min |  
➚ | 6m | 1min | 301min |  
- | 6m | 34min | 335min | oxygen
➚ | 0m | 1min | 336min |  

CNS: 200%
OTU: 415

As you can see dive and deco are exactly the same, as expected (both have the same pO2 in the bottom phase), but the CNS and OTU are very different.

I suspect they are not checking what the oxygen fraction actually is on depth with a PSCR.

### Operating system:

MacOS 10.15.7

### Subsurface version:

4.8.6 latest subsurface from brew.

### Steps to reproduce:

Calculate a PSCR dive and an OC dive where the pO2 is exactly the same on the bottom phase. CNS and OTU will be higher for the PSCR dive

### Current behavior:

Current behavior calculates higher CNS and OTU values for PSCR dives with the same pO2 in the dive.  

### Expected behavior:

The expected behavior is the same CNS and OTU for PSCR and OC dives with the same pO2 values (but different gasses)

### Additional information:

### Mentions:

atdotde, 2021-03-04:
This turns out to be a bigger mess than I anticipated. I don't have time to do it immediately but I have it on my list.

atdotde, 2021-03-11:
Finally found time to sort this out, please see #3216. Note that fO2=31% (OC) gives the same pO2 as  fO2=36% (PSCR) is only true at 18m of depth. The general formula is

fO2 = fO2_gas - (1-fO2_gas) * O2comsumption / (breathing_rate * pscr_ratio)

and so it differs during deco. This explains the small remaining difference ein OTU and CNS values.

PR #3216 (2021-03-12): Use PSCR gas when computing O2 toxicity
Both the calculations for CNS and OTU did not take
into account the pO2 drop when using a PSCR. Furthermore,
there was some unit confusion due to not using internal
units.

Fixes #3203 

Reported-by: arosl

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:
closed 2021-03-12 by atdotde