subsurface/subsurface #3204: Garmin Descent Mk2i MTP - Unable to open raw device

schtibal, 2021-02-25:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Dive import is unable to download dives from Garmin Mk2i. The device is connected successfully via USB in MTP mode, and is visible in Windows Explorer. The device is not visible to be selected from the Subsurface import dialog. Attempting an import anyway appears to connect to the device, but cannot download any dives.

### Operating system:

Windows 10 64bit, English

Garming Descent Mk2i

### Subsurface version:

4.9.10, official release

### Steps to reproduce:

1) Connect Garmin Descent Mk2i via USB, select MTP mode
2) Confirm MTP device is visible in Windows Explorer
3) Open Subsurface, and select Import from Dive Computer
4) Select "Force download of all dives"
4) Select Garmin > Mk2i > Download

### Current behavior:

Error message "No new dives downloaded from dive computer"
If `libdivecomputer` logfile is saved, the following output is generated:
```
Subsurface: v4.9.10, built with libdivecomputer v0.7.0-devel-Subsurface-NG (e58a5866bbd6d12fba7b3482c11d0ae2bed2e1c4)
[0.000000] INFO: Open MTP device
[0.000015] DEBUG: Found Garmin with model 0xcba which is a Mk2/Mk2i

[0.000029] INFO: Read: size=0, data=
[0.001620] DEBUG: Attempting to connect to mtp device
[0.178511] DEBUG: Garmin/mtp: successfully connected with 1 raw devices
[0.193397] DEBUG: Garmin/mtp: unable to open raw device 0
[0.193400] DEBUG: Found 0 files
```

### Expected behavior:

That the device is able to be opened and dives downloaded successfully.

### Additional information:

I have managed to download dives using "MTPDrive" - 3rd party (commercial) software which maps an MTP Device to a drive letter, however this is not a desirable long term solution.

### Mentions:

schtibal, 2021-03-16:
Refer to [mailing list discussion ](https://groups.google.com/g/subsurface-divelog/c/xhpwyfhSsQs)

mikeller, 2026-10-17:
Hi @schtibal,

This should now be fixed: PR #4971 "Windows MSVC build: Add libmtp support" adds native Windows MTP support (via a WPD-based implementation), which is what your Garmin Descent Mk2i needed instead of the MTPDrive workaround.

Closing this as likely fixed, since we don't have an explicit confirmation on a current Windows-MSVC build yet. Please reopen (with a fresh log) if MTP access for your Mk2i still fails after trying a current build.

closed 2026-10-17 by mikeller