subsurface/subsurface #3210: Edit dive in planner gets confused with dive handlers

atdotde, 2021-03-01:
Start with the attached dive, do "edit dive in planner". In the list of waypoints, go to runtime of 16min and control click on the trash bin. This will delete all further waypoints. Then leave the planner with "cancel". 

Back in the dive list, once more do "edit dive in planner". Either it crashes immediately or when your repeat the above delete step, it gets confused with the dive handlers and still shows some of the deleted ones (see screenshot).

I found this in current master, but the problem already existed in 16f86f2f65e73935496e3324bb16f9b8cd4d7953.

[handlers.ssrf.log](https://github.com/subsurface/subsurface/files/6064711/handlers.ssrf.log)

[image]

@bstoeger you did something to the dive point model, any idea?

### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2021-03-01:
I will look into this tomorrow.

bstoeger, 2021-03-02:
Well, a very easy way to make it reliably crash is to CTRL-delete *all* dive points.

atdotde, 2021-03-02:
Oops. 

At least, the last release seems not to be affected.

bstoeger, 2021-03-02:
I fixed the crash with "delete all points" in #3186. It still behaves strangely, but at least I couldn't make it crash easily.

bstoeger, 2021-03-24:
Could you please check if the problem is fixed (by chance) by the rather radical changes in #3186?

atdotde, 2021-03-24:
Will do. Hopefully later tonight, otherwise tomorrow.

atdotde, 2021-03-24:
Indeed, it seems #3186 fixes this.

atdotde, 2021-03-26:
Trying the on the 5.0.0 release crashes subsurface upon opening the dive in the planner the first time. I vote for including #3186 hopefully soon in a 5.0.1 as that version is definitely an improvement to the current state.

closed 2021-06-29 by atdotde