subsurface/subsurface #3214: Event icons missing on mobile

dirkhh, 2021-03-07:
### Describe the issue:

- [x] Bug

### Issue long description:

report from users

### Operating system:

iOS & Android

### Subsurface version:

3.1.3

@atdotde could this be a side effect of the printing changes?

bstoeger, 2021-06-28:
Weirdly, for me (MobileOnDesktop), the event icons are arguably too big. I really need to motivate myself to redo the scaling code. :(

[image]

bstoeger, 2022-03-12:
Can you verify that this is still a problem? Could you give me access to a sample dive?

dirkhh, 2022-03-12:
So with the current app on device I get no events icons. Nothing. No gas changes, no warnings, no alerts.
So I don't think you need a sample dive - any dive with an event should do... most of the dives in the `ssrftest` account have events in them... I know that we filter some of them out, but we should show the alerts and the gas changes...

bstoeger, 2022-03-12:
Can you confirm that you see the event icons on mobile-on-deskop for the same dive log?

What does that debug message say:
```
        if (verbose)
                qDebug("%s DPR: %f metrics: %d %d sz_bigger: %d", __FUNCTION__, dprf, metrics.sz_med, metrics.sz_small, sz_bigger);
```
?

Also, for me things look more harmonious if I remove the mobile-conditional compilation in `divepixmapcache.cpp`.

bstoeger, 2022-03-12:
Moreover, could you please try to replace `DiveEventItem::shouldBeHidden` by a single `return false;` statement?

dirkhh, 2022-03-13:
Yes, I do see the icons in mobile-on-desktop - just not on mobile-on-device.
The problem with the `if (verbose)` in there is that we don't persist switching to verbose mode, and by the time we are able to turn on verbose mode, this constructor has already been run. Dang it.
I'm building a special version for local use where this debug message is printed unconditionally...

And some digging later I realized that I'm an idiot. The icons are bundled incorrectly for on-device.
See #3413

commit 69cb33fd (2022-03-13): build-system: move profile specific icons into separate QRC

This way they are available in both mobile and desktop version.
Without this, the icons weren't shown on iOS and Android.

Fixes #3214
closed 2022-03-13 by dirkhh