subsurface/subsurface #3220: crash opening cloud storage

anpuke, 2021-03-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Program crashes when clicking on open cloud storage. 
See error from eventlog in zipfile: [crash_subsurface.zip](https://github.com/subsurface/subsurface/files/6200942/crash_subsurface.zip)

### Operating system:

Windows 10 Version 20H2 (Build 19042,868)

### Subsurface version:

5.0.0

official release

### Steps to reproduce:

1) file
2) open cloud storage

### Current behavior:

crash
### Expected behavior:

open cloud storage

### Additional information:

### Mentions:

dirkhh, 2021-03-25:
As usual, the MS event info is completely value free.
And unsurprisingly, I can of course happily open cloud storage with the official 5.0.0 Windows binaries (also under Win10).
Not sure what to tell you...

dirkhh, 2021-03-31:
@anpuke Could you test this build https://subsurface-divelog.org/downloads/test/subsurface-5.0.0-5-gfaf349bbdae9.exe
It fixes the issue for me

anpuke, 2021-03-31:
@dirkhh I've installed your build and the error is fixed - cool!

dirkhh, 2021-03-31:
Thanks for the quick check - v5.0.1 is out an contains the fix

closed 2021-03-31 by dirkhh